microsoft/AirSim #1608: Setting up PX4 Software-in-Loop Error

gniwon, 2018-12-05:
[image]

[image]

In July of this year, it worked well without any problems.
But now the above error occurs.

Do you know Why?

jasonhbartlett, 2018-12-07:
I am having the same issue.  This might be a PX4 issue though since I'm getting:
ERROR [px4_daemon] error binding socket

Following instructions here:  [https://github.com/Microsoft/AirSim/blob/master/docs/px4_sitl.md](url) note the following:

$ make posix_sitl_default
Makefile:198:  posix_sitl_default has been deprecated and will be removed, please use px4_sitl_default!

Note recommend to update that doc page to say "make px4_sitl_default"

Anyhow, getting the same error whether I run: 
./build/px4_sitl_default/bin/px4 ./ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t ./test_data
or
make px4_sitl_default none_iris

clovett, 2018-12-08:
are you using the latest version of the PX4 firmware? (v1.8.2) If you have older PX4 firmware, you may need the older px4 command line which was:
```
    ./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris
```

jasonhbartlett, 2018-12-09:
I'm using v1.8.0.  How do I get v1.8.2?

jasonhbartlett, 2018-12-09:
Okay to answer my own question and offer a solution (cause I finally got past this error), recommend checking out this page: [http://dev.px4.io/en/setup/building_px4.html](url)
So once I got a fresh clone of the repo, you can use 'git tag -l' to see a list of available releases.  I used 'git checkout v1.8.2' to get the newest version.  Then you can see your make options by typing 'make list_config_targets' to get a list of all possible configuration targets.  In my case I typed 'make posix_sitl_ekf2 none'  After that it showed me all the config settings and gave me some options to update a bunch sub-modules by typing 'u' ENTER several times.  Then finally PX4 started up!  It still has a couple of errors:
ERROR [parameters] decoder init failed
ERROR [param] importing from 'rootfs/eeprom/parameters' failed (-61)
Not sure if these are show stoppers or not yet.  But at least I have:
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..

Let me know if this helps!

lovettchris, 2018-12-18:
Thanks for the "make posix_sitl_ekf2 none_iris" command line, this works great, and so I updated the docs to reflect that.  PX4 SITL mode now works great, I tested path.py, survey.py, box.py all are working now.

closed 2018-12-18 by madratman

Poyralljack, 2021-12-11:
I was having this issues and my solution is add to head sudo then runs it.

zhangshuaiqing, 2023-10-25:
When I try command 
make px4_sitl_default gazebo
It returns
Makefile:39: *** YOU HAVE TO USE GIT TO DOWNLOAD THIS REPOSITORY. ABORTING.。
What should I do to solve this problem?

yasin-herken, 2023-10-25:
Did you install git for your folder that is in the project

yaasirali32, 2024-03-23:
> Did you install git for your folder that is in the project

I am facing the same problem, i have installed the git, but couldn't be able link it with MATLAB. could you please guide how to solve this issue.