microsoft/AirSim #1609: LandScapeMountain build error in vs2017 [windows]

LeroyLK, 2018-12-06:
I am following the steps from https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md
But when I build the project in VS2017, it cames that

Error		Engine module 'Engine\Source\Runtime\Engine\Engine.Build.cs' should not depend on game module 'LandScape\Source\LandScape\LandScape.Build.cs'	LandScape	D:\Projects\Unreal\LandScape\Intermediate\ProjectFiles\EXEC	1	

PS: I am using window10, UE4.18's Environment (LandScapeMountain)  and VS2017
Thanks for replying in advance!

LeroyLK, 2018-12-06:
And it happens when I am doing the step 9, which is "Press F5 to run. This will start the Unreal Editor. " (in https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md )

LeroyLK, 2018-12-06:
this is my uproject:
{
  "FileVersion": 3,
  "EngineAssociation": "4.18",
  "Category": "",
  "Description": "",
  "Modules": [
    {
      "Name": "Demo1",
      "Type": "Runtime",
      "LoadingPhase": "Default",
      "AdditionalDependencies": [
        "AirSim"
      ]
    }
  ],
  "TargetPlatforms": [
    "MacNoEditor",
    "WindowsNoEditor"
  ],
  "Plugins": [
    {
      "Name": "AirSim",
      "Enabled": true
    }
  ]
}

closed 2018-12-18 by LeroyLK

pengjili, 2019-02-22:
@LeroyLK I am using windows7, and I'm facing the same problem which you met. I did what said it outputs the same result. Do you have any idea how to solve my problem? should I rebuild everything again?

LeroyLK, 2019-02-24:
I am not sure that whether my solution would be fine for you.
I rebuild AirSim by run "build.cmd"., however the eigen3 can't be download by "build.cmd", so I download it mutually.

pengjili, 2019-02-25:
@LeroyLK Thanks for your reply. Actually what I mean is I met the problem of the following line:

Error Engine module 'Engine\Source\Runtime\Engine\Engine.Build.cs' should not depend on game module 'LandScape\Source\LandScape\LandScape.Build.cs'	LandScape

At first I thought it is related to Eigen problem. But I tried all the ways that I know, it just didn't work. So I wanna ask do you solve your problem of this one? If so please tell me your solution, that would be of so big help for me.

LeroyLK, 2019-02-26:
@pengjili 

As I written before, it is the wrong position of eigen3 when I was building AirSim that cause the error:
Error	Engine module 'Engine\Source\Runtime\Engine\Engine.Build.cs' should not depend on game module 'LandScape\Source\LandScape\LandScape.Build.cs'	LandScape	D:\Projects\Unreal\LandScape\Intermediate\ProjectFiles\EXEC	1

So, maybe you should check your build steps.

> Actually, I come across the same problem withe #587
> The root reason is in the step of build AirSim, that is to say "run the build.cmd"
> Because I can't download eigen3 by build.cmd, so I download it mutually, then the mistake come:
> **ONLY the eigen3\Eigen is needed to move into AirLib\deps\eigen3\Eigen**
> To be clearly, the other folders or files in eigen3 is unnecessary!!!