microsoft/AirSim #1611: import setup_path command

Ahmad5112, 2018-12-06:
The Python code in spyder is generating error in setup_path module how it could be sorted
"runfile('C:/Users/Muhammad Ahmed/Desktop/lidar.py', wdir='C:/Users/Muhammad Ahmed/Desktop')
Traceback (most recent call last):

  File "<ipython-input-1-4f7bef62176a>", line 1, in <module>
    runfile('C:/Users/Muhammad Ahmed/Desktop/lidar.py', wdir='C:/Users/Muhammad Ahmed/Desktop')

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/Muhammad Ahmed/Desktop/lidar.py", line 11, in <module>
    import setup_path

madratman, 2018-12-21:
If this was resolved - which I think it is coz of your recent issues with lidar - can we close it?
seems like an issue with your python path.

madratman, 2019-01-08:
@Ahmad5112 ?

closed 2019-01-09 by Ahmad5112