microsoft/AirSim #1616: How to set up AirSim with Custom Unity Environment (Windridge City)?

Gelrant, 2018-12-10:
Hello,

I am trying to figure out setting up a custom environment with AirSim. I have already builded Airsim and tested up with sample Unity projects in the repo. My question is how can i add AirSim plugin to the another Unity environment? Is copying 'Plugins' and 'AirSimAssets' folders are enough?

bugrahanakbulut, 2018-12-11:
Same problem here

Dola47, 2018-12-18:
Same Problem

msb336, 2019-01-03:
Sorry for the radio silence on this.

I've just put in a PR with instructions for baking AirSim into custom unity envs. While it is under review, you can find instructions [here](https://github.com/msb336/AirSim/blob/custom-env-instructions/docs/custom_unity_environments.md).

closed 2019-10-08 by msb336