microsoft/AirSim #1618: Unable to fly drone from a desired location

Ahmad5112, 2018-12-10:
When I start the simulation in the unreal engine my drone appear at the same location while I am trying to change it With ""OriginGeopoint": {
    "Latitude": 4700.641468,
    "Longitude": -1220.140165,
    "Altitude": 12
  } this in settings.json but it doesnt seems to work how can i mange to fly drone from a specific loction .need imidiate help related to issue

lovettchris, 2018-12-18:
Can you try the instructions under "Setting GPS origin" in the [px4_sitl page](https://github.com/Microsoft/AirSim/blob/master/docs/px4_sitl.md)?

Ahmad5112, 2018-12-18:
The problem I was trying to mention was that in Landscape environment when I tried to change the location of multirotor it doesn't. Even when I didn't specify location it randomly appears on a different position on playing simulation. What I want to do is that Multirotor appears on the location that I specified in the setting file
Note : I am not using PX4 controller controlling through simple API

closed 2019-02-19 by madratman

nightduck, 2020-09-08:
Notice to all those who find this issue in the future. The simSetVehiclePose(...) has a glitch where the drone will try to return to the pre-teleport location if takeOffAsync() was called at some point prior. The fix for this is to use simSetVehiclePose and moveToPositionAsync in quick succession, to "lock-in" the new position, like so

    import airsim
    
    client = airsim.MultirotorClient()
    client.confirmConnection()
    drone.enableApiControl(True)
    client.takeoffAsync().join()    

    pose = client.simGetVehiclePose()
    
    # teleport the drone + 10 meters in x-direction
    pose.position.x_val += 10

    client.simSetVehiclePose(pose, True, "PX4")
    client.moveToPositionAsync(pose.position.x_val, pose.position.y_val, pose.position.z_val, 1)