microsoft/AirSim #1624: AttributeError: 'MultirotorClient' object has no attribute 'getLidarData'

Ahmad5112, 2018-12-12:
I am trying to run the code "lidar.py" and getting this error.i have searched for "getLidarData" to include but I did not find it how I can resolve the problem

bmalhigh, 2018-12-12:
@Ahmad5112 -- 
you either didn't update the AirSim plugin in your unreal project with the latest bits OR you didn't update your python client module of AirSim.... most likely you didn't update your python client AirSim module.

Did you install AirSim python module explicitly in the past?

Ahmad5112, 2018-12-12:
no, I didn't install explicitly can you please explain how i can fix it kindly if u tell me in a detail steps it would be really helpful

bmalhigh, 2018-12-12:
1) Make sure you have pulled the latest code from the repo.
2) Build AirSim plugin (build.cmd on windows)
3) Update AirSim plugin in your unreal project
    Essentially copy Unreal\Plugins\AirSim under the plugins folder of your unreal project.
    Are you using Blocks project or your own project?
4) Python module
    a) If you ever installed the AirSim module, update it now by doing
        pip install airsim
    b) If you don't have airsim module installed, just go to PythonClient folder and execute python code

Ahmad5112, 2018-12-12:
can you please give the link to the most updated repository as there are some other forks and they don't work

bmalhigh, 2018-12-12:
I am not sure I understand your comment about multiple forks. The repo we are having this discussion on is the main repo.

git clone https://github.com/Microsoft/AirSim.git

Have you looked at this doc about Python APIs?
https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#python-quickstart

Ahmad5112, 2018-12-12:
Yup I have gone through the said doc before starting and I just have copied the python code for lidar from here and didn't make any change to it well I am cloning again and try again and would let you know if it make any difference

bmalhigh, 2018-12-12:
From the error description (getLidarData not found on client objects), it seems to be stale bits issue. My hunch is stale bits on the python client side.

Can you check using "pip list" to make sure an older AirSim python module is not installed on your machine?

Ahmad5112, 2018-12-12:
AirSim 1.2.3 is installed as i checked and got the following list

[image]

bmalhigh, 2018-12-12:
Right... I am assuming you installed airsim python module in the past.

Check to understand what bits are used by that module.
From a python shell, execute
    import airsim
    airsim.__file__
That should show you where the install module is coming from

Ahmad5112, 2018-12-12:
on running this in Anaconda i got the following
runfile('C:/Users/Muhammad Ahmed/Desktop/Simulation/untitled1.py', wdir='C:/Users/Muhammad Ahmed/Desktop/Simulation')
Reloaded modules: airsim, airsim.client, airsim.utils, airsim.types, msgpackrpc, msgpackrpc._version, msgpackrpc.loop, msgpackrpc.client, msgpackrpc.session, msgpackrpc.message, msgpackrpc.future, msgpackrpc.error, msgpackrpc.transport, msgpackrpc.transport.tcp, tornado.iostream, tornado.netutil, msgpackrpc.compat, msgpackrpc.server, msgpackrpc.address

bmalhigh, 2018-12-12:
Are you able to uninstall airsim package?
    `pip uninstall airsim`

Ahmad5112, 2018-12-12:
I have just finished up with building AirSim again and also uninstalled it from pip directory what to do next?

bmalhigh, 2018-12-12:
Two options:
1) go ahead and install the new airsim bits from \PythonClient directory
    `pip install airsim`
   OR
2) Do what the files under \PythonClient\Car and \PythonClient\Multirotor are doing.
    They import a file called setup_path.py (present locally in those folders). That file can use the airsim directly from your clone without requiring install of airsim package.

If confusing, just use option#1 and try.

Ahmad5112, 2018-12-12:
After doing All the fixes now i am getting This

  File "C:\ProgramData\Anaconda3\lib\site-packages\airsim\types.py", line 17, in <dictcomp>
    obj.__dict__ = { k : (v if not isinstance(v, dict) else getattr(getattr(obj, k).__class__, "from_msgpack")(v)) for k, v in encoded.items()}

AttributeError: 'LidarData' object has no attribute 'pose'

bmalhigh, 2018-12-12:
In the types.py file, check the LidarData class defined at the very bottom.
Do you see the pose property in that file?

`class LidarData(MsgpackMixin):
    point_cloud = 0.0
    time_stamp = np.uint64(0)
    **pose = Pose()**`

Ahmad5112, 2018-12-12:
yes it has the property 
does**pose = Pose()** this staric have some meaning as these are missing

Ahmad5112, 2018-12-12:
I am running code from Spyder is there any other way to run the code?

bmalhigh, 2018-12-12:
I am assuming your using the Blocks env from \Unreal\Environments\Blocks\

In this file
     \Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\CommonStructs.hpp

Do you see the Pose property on the LidarData struct?
`
struct LidarData {

    TTimePoint time_stamp = 0;
    vector<real_T> point_cloud;
    Pose pose;

    LidarData()
    {}
};
`

Ahmad5112, 2018-12-12:
yes it is there
struct LidarData {

    TTimePoint time_stamp = 0;
    vector<real_T> point_cloud;
    Pose pose;

    LidarData()
    {}
};

}}

bmalhigh, 2018-12-12:
@sytelus -- any ideas on how to debug this further? It seems like an issue due to stale bits on the python client side... but I am not sure what to look at next.

james-mchugh, 2018-12-19:
I am also experiencing similar issues. When I try to access the pose attribute, it throws:

> AttributeError:  'LidarData' object has no attribute 'pose'

james-mchugh, 2018-12-19:
Actually, I see what may be causing the issue. If you are using any of the releases, `LidarData` does not have the `pose` object. It is only on master. This is what is causing the issue on my end at least...

ma8tsch, 2019-02-24:
Unfortunatly I get the same error:

AttributeError: 'LidarData' object has no attribute 'pose'

Does anyone know how to fix this issue? (My airsim branch is on master and there is nothing to pull)

orangeccc, 2019-04-23:
I fixed this by edit the  file  \Anaconda3\Lib\site-packages\airsim\types.py.  
`class LidarData(MsgpackMixin):
    pose = Pose()
    point_cloud = 0.0
    time_stamp = np.uint64(0)`
It seems that pose is missing.

jaymefosa, 2019-04-30:
@orangeccc This worked for me, thank you

madratman, 2019-04-30:
I think everyone here just needs to do this instead of `pip install airsim`

```
cd PythonClient;
python setup.py install  # use sudo or if you get a permissions error.  
```

madratman, 2019-05-14:
We have updated the pypi package here https://pypi.org/project/airsim/ and a `$ pip install airsim --upgrade --user`, or `$ sudo pip install airsim --upgrade` should do the trick

closed 2019-05-14 by madratman

majingchunn, 2024-05-04:
> 我通过编辑文件\Anaconda3\Lib\site-packages\airsim\types.py 解决了这个问题。
> 
> 类 LidarData（MsgpackMixin）：
> 
> ```
> pose = Pose()
> point_cloud = 0.0
> time_stamp = np.uint64(0)
> ```
> 
> 似乎缺少了。` pose = Pose()`

这似乎对我没用