microsoft/AirSim #1627: Point cloud generation in stereo mode

shmm91, 2018-12-13:
I am using Airsim to generate stereo images, then use it to compute point cloud. But I am confused with camera projection matrix, it's different with Opencv or ROS. Also, I read issues about the related topics which make me more confused. 
This is my setting parameters:
 "Width": 640,
 "Height": 480,
 "FOV_Degrees": 90
This is my python file:

import setup_path
import airsim
import cv2
import rospy
import numpy as np
from sensor_msgs.msg import PointCloud2
from sensor_msgs.msg import PointField

# camera parameters
Fx = 320
Fy = 320
Cx = 320
Cy = 240
Tx = 0.25

projectionMatrix = np.array([[1, 0, 0, -Cx],
                                            [0, 1, 0, -Cy],
                                            [0, 0, 0, Fx],
                                            [0, 0, -1/Tx, 0]])

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

pcl_pub = rospy.Publisher("airsim/pointcloud2", PointCloud2, queue_size=1)
rospy.init_node('pointcloud', anonymous=True)
rate = rospy.Rate(30) # 30hz

# quadrotor takeoff
landed = client.getMultirotorState().landed_state
if landed == airsim.LandedState.Landed:
    print("taking off...")
    client.takeoffAsync().join()

while not rospy.is_shutdown():

    dispImage = client.simGetImage("2", airsim.ImageType.DisparityNormalized)
    png = cv2.imdecode(np.frombuffer(dispImage, np.uint8), cv2.IMREAD_UNCHANGED)
    gray = cv2.cvtColor(png, cv2.COLOR_BGR2GRAY)
    Image3D = cv2.reprojectImageTo3D(gray, projectionMatrix)

    pclmsg = PointCloud2()
    pclmsg.header.stamp = rospy.Time.now()
    pclmsg.header.frame_id = "pointcloud"
    pclmsg.width = Image3D.shape[1]
    pclmsg.height = Image3D.shape[0]
    pclmsg.fields = [PointField('x',0,PointField.FLOAT32,1),
                             PointField('y',4,PointField.FLOAT32,1),
                             PointField('z',8,PointField.FLOAT32,1)]
    pclmsg.is_bigendian = False
    pclmsg.is_dense = False
    pclmsg.point_step = 12
    pclmsg.row_step = 12*Image3D.shape[1]
    pclmsg.data = np.asarray(Image3D,np.float32).tostring()

    pcl_pub.publish(pclmsg)

    # sleep until next cycle
    rate.sleep()

This is my result in RVIZ and raw image, the point cloud seems reversed in y axis.
[image]
[image]

chaitanyantr, 2019-02-12:
What do you mean by stereo mode...!

Your getting images camera id as 2....so you mean camera 2 is stereo ?

Multi-rotor have ---- front_center, front_right, front_left, bottom_center and back_center .-----
Which one is stereo in above cameras..?

dnychennnn, 2020-06-22:
Hi, 
have you figured out how to correctly use the reprojectImageTo3D function? I'm trying to get the point cloud from DepthPerspective image real time(iteratively), but somehow the results doesn't seem very promising for me.

closed 2021-11-30 by jonyMarino