microsoft/AirSim #1629: lidar sensor reading related querry

Ahmad5112, 2018-12-13:
I am running the drone. lidar code to understand the lidar functionality and I got this reading in return I am unable to understand the output how can I interpret it.If somebody can please help me to understand AS I need to use lidar in my Collision Avoidance algorithm  
Reading 1: time_stamp: 1544726195660222464 number_of_points: 93
                lidar position: <Vector3r> {   'x_val': -9.906064987182617,
    'y_val': 9.910856246948242,
    'z_val': -10.954754829406738}
                lidar orientation: <Quaternionr> {   'w_val': 0.9971162676811218,
    'x_val': -0.05290545895695686,
    'y_val': -0.05430632457137108,
    'z_val': -0.0033124182373285294}

bmalhigh, 2018-12-17:
Hi,

1) "timestamp" represent when the lidar capture was done.
    See code here, 
    https://github.com/Microsoft/AirSim/blob/master/AirLib/include/sensors/lidar/LidarSimple.hpp 
          `output.time_stamp = clock()->nowNanos();`
2) "number_of_points" indicates how many points were hit in the laser scan at that timestamp.
3) "lidar position" is the position of the lidar (in NED) in the vehicle-inertial-frame (i.e., origin being the start of the vehicle). "lidar orientation" is the also in the vehicle-inertial-frame.
4) By default, the lidar point-cloud data is in "VehicleInertialFrame" (origin being the vehicle start position). But you can use AirSimSettings "DataFrame" to specify "SensorLocalFrame" and then the lidar points will in the lidar frame (i.e., the origin being the pose of the lidar)

https://github.com/Microsoft/AirSim/blob/master/docs/lidar.md

Ahmad5112, 2018-12-17:
well I have tried with lidar local frame settings  but the problem is I am unable to interpret the data in return that is also with label lidar position and orientation can you please explain  what does these points describe how can I measure the distance of the obstacle from that point I got in return
Here are the points I got with that setting that you mention
wherever vehicle position I have specified point to take lidar data is "self.client.moveToPositionAsync(5  , -5, -6, 5).join()"
 Reading 1: time_stamp: 1545057439506942464 number_of_points: 1110
                lidar position: <Vector3r> {   'x_val': 4.856394290924072,
    'y_val': -4.846883773803711,
    'z_val': -6.915310859680176}
                lidar orientation: <Quaternionr> {   'w_val': 0.9983316659927368,
    'x_val': 0.04017558693885803,
    'y_val': 0.04134482517838478,
    'z_val': -0.00323234754614532}
        Reading 2: time_stamp: 1545057444301044736 number_of_points: 1110
                lidar position: <Vector3r> {   'x_val': 9.211358070373535,
    'y_val': -9.238350868225098,
    'z_val': -6.997963905334473}
                lidar orientation: <Quaternionr> {   'w_val': 0.9999614357948303,
    'x_val': 0.0037750648334622383,
    'y_val': 0.003786010667681694,
    'z_val': -0.006971627939492464}
        Reading 3: time_stamp: 1545057448993144832 number_of_points: 1110
                lidar position: <Vector3r> {   'x_val': 9.35843276977539,
    'y_val': -9.38703441619873,
    'z_val': -6.997913360595703}
                lidar orientation: <Quaternionr> {   'w_val': 0.9999756813049316,
    'x_val': 0.000132297893287614,
    'y_val': 0.00013272764044813812,
    'z_val': -0.006972187664359808}
        Reading 4: time_stamp: 1545057453685244928 number_of_points: 1110
                lidar position: <Vector3r> {   'x_val': 9.363611221313477,
    'y_val': -9.392269134521484,
    'z_val': -6.997924327850342}
                lidar orientation: <Quaternionr> {   'w_val': 0.9999756813049316,
    'x_val': 4.63655896965065e-06,
    'y_val': 4.65337461719173e-06,
    'z_val': -0.006972187664359808}

bmalhigh, 2018-12-18:
Ahmad, I am not sure I totally understand your question.

There are few things in the data,
1) Position and orientation of the lidar itself, and,
2) Point cloud representing the points around the lidar that were hit by the lidar scan.

If you are looking for the distance of the obstacles from the lidar, then you would need to look into pointcloud data. See 

https://github.com/Microsoft/AirSim/blob/master/PythonClient/multirotor/drone_lidar.py
`
 def parse_lidarData(self, data):

        # reshape array of floats to array of [X,Y,Z]
        points = numpy.array(data.point_cloud, dtype=numpy.dtype('f4'))
        points = numpy.reshape(points, (int(points.shape[0]/3), 3))
       
        return points
`

The points returned by this method represent the (x,y,z) of the points that were hit by the lidar scan.

One thing to note though is that the point-cloud represents the points that were hit. It doesn't represent what obstacle that point is on.

vishal-prgmr, 2019-04-24:
@bmalhigh , I feel the word "VehicleInertialFrame" is confusion as it sounds like a mix between the VehicleBody Frame and the Global Frame, cause here you refer to the VehicleInertialFrame as the start position of the car. At the first time stamp, this would be the same vehicle frame, but as time progresses, this is the Global Frame (because you say the origin is the start position of the car). 

Kindly clarify. Thank You

yigitkaraduman, 2020-01-30:
If you meant visiulazing the lidar data you can use pptk.

pip install pptk

import pptk

in car_lidar.py module,
points = self.parse_lidarData(lidarData)                       
 v = pptk.viewer(points)

closed 2021-10-01 by jonyMarino