microsoft/AirSim #1630: Need Guidance Related to USing Lidar

Ahmad5112, 2018-12-13:
I need to use Lidar Sensor in my Collision Avoidance Algorithm. I actually need a measurement of lidar in term of distance from obstacle how can I do this task in AirSim.

bmalhigh, 2018-12-17:
Ahmad, per the facebook conversation -- 

1) You can configure the Lidar to point in a particular horizontal and vertical FOV and configure it to return data in "SensorLocalFrame"..... and then use getLidarData API to get the data.
2) In "SensorLocalFrame" config, the PointCloud returned by getLidarData represents the position of the points (x,y,z in NED) that were hit with the Lidar being the origin.
3)  The distance of a point (x,y,z) being hit should be sqrt((x)^2 + (y)^2 + (z)^2) from the lidar... but remember that lidar doesn't lock its laser on the obstacle.

Ahmad5112, 2018-12-17:
Sorry bmalhigh for being too sticky, but the problem is I am still not able to understand and I have mentioned my confusion in detail in the previous thread on which you have commented

Ahmad5112, 2018-12-25:
Bmalhigh I have followed your instruction for settings but the almost same point is returned by "get lidar data " API as lidar reading .what does this mean

jonyMarino, 2020-08-17:
I suggest to read the latest documentation on this subject: https://microsoft.github.io/AirSim/lidar/

closed 2020-08-17 by jonyMarino