microsoft/AirSim #1632: Camera Translation via python API

zsfVishnu, 2018-12-14:
For my drone simulation project, i wanted to translate the front right and front left cameras towards the respective sides. I saw the orientation change commands but there is nothing relating to the translation. Is there any other way to do this without adding extra cameras. Also, i need the solution via editing the Python API and not the json file (i saw a similar issue solution based on this)

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

rajat2004, 2020-08-20:
The new [`simSetCameraPose` API](https://github.com/microsoft/AirSim/blob/master/docs/image_apis.md#camera-apis) allows translation of the camera along with the rotation

jonyMarino, 2020-08-20:
Thank you very much @rajat2004 for your help. I still have to get familiar with the apis

closed 2020-08-20 by jonyMarino

rajat2004, 2020-08-21:
Glad to help @jonyMarino! Yeah, the APIs are not very well documented, having a reference doc will be good rather than needing to dive into the source code every time

sandilyasg, 2024-01-14:
@rajat2004 @jonyMarino @zsfVishnu to use the [simSetCameraPose API](https://github.com/microsoft/AirSim/blob/master/docs/image_apis.md#camera-apis), do we have to be in computer vision mode in settings.json, or can it also run in multirotor mode?