microsoft/AirSim #1634: Setting GPS origin (not a dup)

jasonhbartlett, 2018-12-14:
Windows 10 Pro
VC++ and 8.1 SDK
Unreal 4.18
PX4 release v1.82 SITL running in BashOnWindows

I can't seem to get the GPS position to change to anything other than the default Microsoft location.  Settings file looks like this:
```yaml
{
  "SettingsVersion": 1.2,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "SitlPort": 14540,
      "UdpPort": 14560
    },
    "OriginGeopoint": {
      "Latitude": 22.915464,
      "Longitude": -59.975619,
      "Altitude": 20
    }
  }
}
```
(Note: GPS location changed for privacy.)
Have changed the settings in the PX4 terminal:
[image]

But when I play the Blocks environment and then open QGroundControl it always brings me to the Microsoft campus!

Are those default GPS coordinates still hardcoded into the program somewhere and my Settings.json OrginGeopoint being ignored?

jasonhbartlett, 2018-12-17:
FYI - Here are some quick videos I made - Combining AirSim Unity with [Mapbox](www.mapbox.com/unity) Would really like to synchronize the GPS between the two.

[Flying drone with controller through cityscape provided by Mapbox](https://youtu.be/ESYuNAMAboY) 

[PX4 SITL through cityscape provided by Mapbox.  How can I sync GPS between these?](https://youtu.be/YmhckyUpo7k )

[Flying drone over Washington, DC satellite imagery](https://youtu.be/oClJXJpBIAE)

lovettchris, 2018-12-18:
You could try setting these PX4 parameters to the GPS location your map expects to start with.
```
param set LPE_LAT 47.641468
param set LPE_LON -122.140165
```

jasonhbartlett, 2018-12-18:
@lovettchris I think I'm missing something fundamental here.  (By "map", do you mean AirSim?) I want my map to "expect to start" somewhere else besides that location.  So for example I want the AirSim drone starting location to be at my house so I would A.) Use my GPS coordinates in the Setting.json file and B.) Set PX4 parameters to use my GPS coordinates as well.  Then when I fire up QGroundControl I should see a map of my neighborhood.  But that's not happening.  I still see Microsoft campus location.  How do I change that?

lovettchris, 2018-12-18:
Oh, perhaps the problem is you have the "OriginGeopoint" inside the section for "Vehicles", try promoting it up a level like this:
```json
{
  "SettingsVersion": 1.2,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "SitlPort": 14540,
      "UdpPort": 14560
    },
  }
  "OriginGeopoint": {
    "Latitude": 22.915464,
    "Longitude": -59.975619,
     "Altitude": 20
    }
}
```

jasonhbartlett, 2018-12-18:
@lovettchris 
Ahh... that was it!  Simple.  Works for Unity & Unreal.  Thanks so much for your help!

closed 2018-12-18 by jasonhbartlett

lovettchris, 2018-12-18:
And here's how I double checked using SITL px4 app:
[image]

thejeshk, 2020-03-10:
Hi @lovettchris, 

Windows 10
Unreal 4.22

My Vehicle is Car
Is there any way to set GPS info to the car and spawn the car at specific GPS position? 

I have my custom environment (which is duplicated real test site) 

Thank you for the help :)