microsoft/AirSim #1636: Unreal Block Environment VS 2017

shubham9436, 2018-12-17:
I have followed all the steps.
First I cloned the **AirSim repo** (latest) and using Developer Command Prompt for VS 2017, I have **built the Airsim repo** successfully.
Then I want to setup the Unreal Block environment.
I clicked on **update_from_git.bat** file and now **Blocks.sln** file is appearing.
I set Blocks as startup project, DebugGame Editor and Win64.
Now when I run F5, few errors pop up.
I have attached the screenshots and output of build.

[image]
[image]
[build_output.txt](https://github.com/Microsoft/AirSim/files/2684400/build_output.txt)

Please give the solutions.
I am using Unreal Engine 4.1, Windows 10, VS 2017 and AirSim(latest pulled repo).

openweb, 2018-12-17:
same error and question, thanks.

openweb, 2018-12-17:
When using UE 4.18, problem is solved :)

shubham9436, 2018-12-17:
> If used UE 4.18, problem is solved :)

Thanks for the reply
I'll try and let you know...

bmalhigh, 2018-12-17:
U 4.18 is the supported version at this time.