microsoft/AirSim #1643: MoveToPositionAsync: Offset in Unreal position

majouda7, 2018-12-18:
Hello,

I am using Python Client API  (move* API) to  build a path planner algorithm. I am having an issue with moveToPosition API call, after giving the command to move to a certain position I am noticing a huge offset in the drone location in unreal engine. For instance, if I send the drone to the  way point (-5,0,-3) (client.moveToPositionAsync(-5, 0, -3, 5).join()) with a velocity of  5m/s, the drone actually goes to (-10,0,-4) in unreal environment (Ned frame of the drone). I have also noticed that this error goes up with velocity and it is a kind of random (not always the same offset for the same way point).   Moreover, it seems that AirSim is not aware about this offset in unreal because when I call client.simGetVehiclePose().position it returns exactly the same position I requested at the beginning (-5,0,-3).   This is very undesirable especially in a path planning context where a reasonable precision in the environment is required to avoid obstacles. Is there a way to fix that ?

Thanks,

saihv, 2019-01-15:
I think I have the same problem. I am trying to control multiple SimpleFlight drones with the Python API, and I am encountering errors in how MoveToPositionAsync actually moves the drones. For instance, this is a set of commands I gave the drone:

```
f1 = client.moveToPositionAsync(5, 0, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(-5, 0, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(-5, -10, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(5, -10, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(5, 0, -8, 3, vehicle_name="Drone1")
```

My world is properly aligned, so this should make the drone fly in a perfect square. Instead, this is what the drone does (plotted from the airsim_rec log):

[image]

Unless I am doing/understanding something wrong, this seems to be a pretty major issue.

madratman, 2019-02-19:
let's track this in 1685

closed 2019-02-19 by madratman