microsoft/AirSim #1644: Drive Car in Windridge City

Dola47, 2018-12-18:
Hello,

I installed Windridge City. I would like to be able to drive the car in the environment.

I checked and found that the Car is defined in AirSimAssets I tried to copy some folder to Windridge City Assets but that was not helpful.

When I press the play mode button in SimModeSelector and choose Car Mode, CarDemo is launched which allow the car to be driven. I would like to do the same but in the new Windridge Environment.

Any Suggestions how to do so?

vogelo, 2018-12-19:
I managed to do this, steps were

1. Copying car and camera from CarDemo to Windridge Scene
2. Changing Position of car and camera to the camera of windridge scene
3. removing the old windridge camera
4. adding the windridge scene in build settings
5. Hacking the script which starts the scene after pressing the Car mode button to start windridge scene
6. Re-creating the link from the camera to Car/LookAt for the Follow script

Dola47, 2018-12-19:
@vogelo 

As Far as I can understand:

1- You just copied them from CarDemo inside Unity and paste them back in Windridge scene also from inside Unity.

2- You adjusted their position in the list of Windridge as they were in CarDemo.

3- You replaced the new camera from CarDemo by the Camera that was by default defined in Windridge Scene.

4- Okay.

Point no. 5 & 6 are unclear to me how they are done. Please, try to illustrate them in more detail. I do not understand how to hack this script and what is actually this script. 

Instead I copied everything inside Script folder of AirSimAssets and paste them back in to Windridge City Assets and I got the error shown in the attached photo. 

I think if I done step no. 5 & 6 as you already done I will have the car moving. 

Thanks.
[image]

msb336, 2019-01-08:
Hello,

you can find instructions for adding airsim to unity environments [here](https://github.com/Microsoft/AirSim/blob/master/docs/custom_unity_environments.md)