microsoft/AirSim #1645: Lidar: HorizontalFOV not working

edualvarado, 2018-12-19:
OS: Ubuntu 16.04
AirSim 1.2

I have a Lidar sensor in my vehicle. I want to setup the horizontal range, but it is not working. I want to specify an horizontal field of view of 270 degrees (-45, 225). To do so, in the settings.json file, I specify the following (for a 2D Lidar in this case):

```
    "Lidar1": {
         "SensorType": 6,
         "Enabled" : true,
         "NumberOfChannels": 1,
	 "Range": 25,
         "RotationsPerSecond": 10,
         "PointsPerSecond": 10000,
         "X": 0, "Y": 0, "Z": -0.15,
         "Roll": 0, "Pitch": 0, "Yaw" : 0,
	 "VerticalFOVUpper": 0,
	 "VerticalFOVLower": 0,
	 "HorizontalFOVStart": -45,
	 "HorizontalFOVEnd": 225,
         "DrawDebugPoints": true
       }
```

The vertical FOV works fine when the Lidar has 2 or more channels (for 2D Lidars with 1 channel it doesn't make any difference). But the horizontal FOV does not work in any case. The points are always in a 360 degrees aperture. It doesn't natter if the Lidar has 1 channel or more. I think it could be a bug. Any thoughts?

Thanks,

RVBldr, 2019-01-08:
We'll take a look, this should be working.

bmalhigh, 2019-01-09:
Horizontal-fov was implemented a bit later. Can you please pull in the latest code from repo & try?

madratman, 2019-01-23:
closing as this is resolved. please re-open/comment if you can't get hfov working

closed 2019-01-23 by madratman

shashankyld, 2020-12-29:
I have the same problem, cant use 2D lidar in Airsim! Any suggestions?