microsoft/AirSim #1646: AirSim Unity ignoring resolution settings in settings.json

vogelo, 2018-12-19:
I am using AirSim on Unity (based on UnityDemo), on the Windridge City environment, on Linux. In principle, it works fine, up to the important step of saving images.

I would like to capture images using the python API (scene and depth, but let's go with scene first). However, when I do so using

`responses = client.simGetImages([airsim.ImageRequest("0", airsim.ImageType.Scene)])`

I always get 256x144 images. I verified airsim.ImageType.Scene to be 0.

Now, I read that I can change this by changing defaults in settings.json, which is located in $HOME/Documents/AirSim/ and actually IS loaded (I confirmed in the log), which contains

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 640,
          "Height": 480,
          "FOV_Degrees": 60,
          "AutoExposureSpeed": 100,
          "MotionBlurAmount": 0
        }
    ]
  },
  "SimMode": "Car"
}

```

However, this has no effect. Images remain to be 256x144, i.e. either settings.json is ignored completely, or the Width/Height settings are ignored.

Any idea what I'm doing wrong?

msb336, 2019-01-08:
@josedeo 
Any comment about this one?

adrshm91, 2019-01-13:
I had the same problem. There is a problem with the settings.json file provided with https://github.com/Microsoft/AirSim/blob/master/docs/settings.md. It needs to be set as

> {
 "SimMode" : "Car", 
 "SettingsVersion" : 1.2, 
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "CaptureSettings": [
 	{
 		 "Width": 640,
        "Height": 480
 	}
 ],
}

It works for me now. Looks like there is no field called CameraDefaults in the script AirSimSettings.cs .  Could someone please update the default settings file in the readme?

closed 2019-01-24 by msb336

vishal-prgmr, 2019-04-23:
@adrshm91 , Tried removing CameraDefaults. But still the resolution doesnt change.