microsoft/AirSim #1649: getSimImages() broken on Unity

vogelo, 2018-12-20:
When requesting images using getSimImages() in Unity on Linux using the Python API, I get the following results based on stated requests. Version is master/latest.

(requested) -> (returned) --- (evaluation)
Scene -> Scene --- (good)
DepthPlanner,pixels_as_float = True -> Scene,pixels_as_float = False  --- (Terrible)
DepthPerspective,pixels_as_float = True -> Scene,pixels_as_float = False  --- (Terrible)
DepthVis -> Segmentation --- (what???)
Segmentation -> Segmentation --- (good)

If I, however, request DepthVis through getSimImage() (without the s, which I cannot use to get depth, which I really need), I get DepthVis. So I assume there must be an issue with the getSimImage (or ImageRequest) interface.

vogelo, 2018-12-20:
More information: Actually, getSimImage() only seems to work when called at least twice. The first call returns the wrong image.

howard-mahe, 2019-04-04:
Hi, I am also facing a lag of 1 image **per camera** using either `getSimImage()` or `getSimImages()`.

Example: 
```python
# "0" is front camera, "1" is rear camera
responses = client.simGetImages([
   airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False), # dummy because "0" never used before
   airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False), # >> front + Segmentation
   airsim.ImageRequest("0", airsim.ImageType.DepthVis    , False, False), # >> front + Segmentation
   airsim.ImageRequest("0", airsim.ImageType.DepthVis    , False, False), # >> front + DepthVis
   airsim.ImageRequest("0", airsim.ImageType.Scene       , False, False), # >> front + DepthVis
   airsim.ImageRequest("1", airsim.ImageType.Scene       , False, False), # >> dummy because "1" never used before
   airsim.ImageRequest("0", airsim.ImageType.DepthVis    , False, False), # >> front + Scene
   airsim.ImageRequest("1", airsim.ImageType.DepthVis    , False, False), # >>  rear + Scene
   airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False), # >> front + DepthVis
   airsim.ImageRequest("1", airsim.ImageType.Segmentation, False, False), # >>  rear + DepthVis 
])
```

ClarkZha, 2021-02-27:
Bumped into same problem.
I don't this this bug is fixed yet (as of Feb.2021).

Dug a little bit into the problem and here is what I found:

1. DepthPlanner and DepthPerspective not working: 
In CameraFilterScript.cs:
[image]
Apparently `DepthPlanner` and  `DepthPerspective` is not setup for Unity. When you request them, the case becomes default, which is `Scene`.

2. The render texture setup seems problematic:
In DataCaptureScript.cs:
[image]
[image]
`SetUpRenderType` is always called before `SetUpRenderTextureForCapture` when an image is captured. As a result, the `SetUpRenderTextureForCapture` won't do anything as it returns in the if clause. This seems extremely fishy as `SetUpRenderTextureForCapture` is always called for nothing. I think this is related to the error of capturing wrong first image and the filter gets updated, but the image was not re-rendered, so the the captured image is still wrong. 

Very new to the Unity world. I tried to fix the problem by flip the order the functions are called, but I end up getting pictures that are completely black. Would love to hear some expert's opinion on this.