microsoft/AirSim #1650: Hi,

samby707, 2018-12-20:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

samby707, 2019-01-02:
When I'm running build.cmd from x64 native tools visual studio command prompt for AirSim library, it is getting built successfully but the Airsim.uplugin file is not getting generated.

sbkybhy, 2019-09-27:
Hello, I have the same problem. How did you solve this problem

closed 2020-04-14 by madratman