microsoft/AirSim #1651: Callback function for collision

koriavinash1, 2018-12-20:
Are there any callback functions which are triggered during collision for python airsim client ?

koriavinash1, 2018-12-20:
@NeedHealth simGetCollisionInfo(.) returns only when it's called, I wanted to know if there are any functions which are automatically triggered when drone/car collides with obstacle and can be used a callback in simMoveByVelocity(.) or simMoveToPosition(.)

koriavinash1, 2018-12-27:
@AloshkaD the method you suggested works but in your approach the condition is checked only once in a loop. If you have callback function, action terninates (or do anything as defined in callback function) whenever drone collides, you need not wait till the exicution of above condition.

AloshkaD, 2018-12-30:
That's right. I don't worry about that because In my application I use it in an openai gym env and it's part of my `step` so I get to check collision in a while loop during each taken step.

madratman, 2019-01-23:
You can implement a "timer callback" (misnomer inspired by [ROS](https://wiki.ros.org/roscpp/Overview/Timers)). Something like [this](https://stackoverflow.com/questions/8600161/executing-periodic-actions-in-python)

closed 2019-01-23 by madratman