microsoft/AirSim #1652: Snap Package Support (Feature Request)

mikeroyal, 2018-12-21:
I was thinking the install and setup could be improved for Ubuntu and other Linux distros by having a [snap package](https://snapcraft.io/) version. If it's not a priority would you mind if I made the snap as a contribution?

[Crafting your first snap](https://tutorials.ubuntu.com/tutorial/create-your-first-snap#0)

[Build and test snap](https://snapcraft.io/build)

[Getting Started snap template](https://github.com/snapcrafters/fork-and-rename-me)

madratman, 2018-12-21:
Yea, a PR would be helpful and welcome. Please go ahead, and ask any questions if you're stuck

jonyMarino, 2020-09-24:
Hi! It seems that the Ubuntu community is leaving snap, so we are going to stick with only the publication of binaries.

closed 2020-09-24 by jonyMarino