microsoft/AirSim #1657: IncrementFlushOnDraw is not a member of FSceneViewport.

smilefacehh, 2018-12-23:
Does anybody know how to solve this? I can't find the IncrementFlushOnDraw  method in class FSceneViewport. Thx.

MathieuCesar, 2018-12-26:
I realized I was using the latest version of the Unreal Engine (4.21) after installing using the default command:
git clone https://github.com/EpicGames/UnrealEngine.git
Instead, using the recommended version of the Unreal Engine (4.18) solved the problem:
git clone -b 4.18 https://github.com/EpicGames/UnrealEngine.git

madratman, 2019-01-23:
We had a few fixes for UE 4.21 last week in https://github.com/Microsoft/AirSim/commit/a179075a2a546537098c11cd1dcc1e8ff6589612. Might be worth trying again

hardsv, 2019-01-24:
I am facing the same issue ....using ue4.21.2 and Airsim latest version... downloaded it today.

1>------ Build started: Project: Blocks, Configuration: DebugGame_Editor x64 ------
1>Creating makefile for BlocksEditor (changes to module files)
1>While compiling D:\work\AirSim\AirSim\Unreal\Environments\Blocks\Intermediate\Build\BuildRules\BlocksModuleRules.dll:
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(50,17): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(51,17): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(55,17): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(61,17): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(83,9): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(84,9): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(85,9): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>d:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(154,9): warning CS0618: 'UnrealBuildTool.ModuleRules.Definitions' is obsolete: 'The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
1>Using Visual Studio 2017 14.16.27023 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023) and Windows 10.0.17763.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>Building 4 actions with 4 processes...
1>  [1/4] Module.AirSim.cpp
1>D:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirBlueprintLib.cpp(127): error C2039: 'IncrementFlushOnDraw': is not a member of 'FSceneViewport'
1>  C:\Program Files\Epic Games\UE_4.21\Engine\Source\Runtime\Engine\Public\Slate/SceneViewport.h(32): note: see declaration of 'FSceneViewport'
1>D:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirBlueprintLib.cpp(134): error C2039: 'DecrementFlushOnDraw': is not a member of 'FSceneViewport'
1>  C:\Program Files\Epic Games\UE_4.21\Engine\Source\Runtime\Engine\Public\Slate/SceneViewport.h(32): note: see declaration of 'FSceneViewport'
1>UnrealBuildTool : error : UBT ERROR: Failed to produce item: D:\work\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\AirSim\UE4Editor-AirSim-Win64-DebugGame.lib
1>                        (see ../Programs/UnrealBuildTool/Log.txt for full exception trace)
1>Total build time: 133.42 seconds (Parallel executor: 0.00 seconds)
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.21\Engine\Build\BatchFiles\Build.bat" BlocksEditor Win64 DebugGame "D:\work\AirSim\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "Blocks.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Any help would be appreciated.. thanks

hardsv, 2019-01-28:
thanks @sreejithkrishnanr ....it sounds like we are hacking the hack. I am going to try your suggestions and lets see if affect the my project. But I feel eventually I have to shift to 4.18.

coldwarrior5, 2019-02-07:
Apparently, according to the Unreal Engine changelog on GitHub, they removed the functionality because it was unnecessary.

Change 4093867 by Andrew.Rodham

Sequencer: Added the ability to implement custom capture protocols for movie scene captures
          ...
	  - Removed Flush On Draw functionality from viewports and frame grabber since it is unnecessary.

Therefore you are unable to force FlushRenderingCommands(); in the UnrealClient.cpp through Increment and DecrementFlushOnDraw

hardsv, 2019-02-08:
Thanks @coldwarrior5 for the explanation. 
I am able to compile after I comment out those lines. Till now haven't seen any problem working with maps. 

> Apparently, according to the Unreal Engine changelog on GitHub, they removed the functionality because it was unnecessary.
>
> Change 4093867 by Andrew.Rodham
>
> Sequencer: Added the ability to implement [...]

coldwarrior5, 2019-02-08:
The only thing that needs to be validated is the framerate of sensor data when the NoDisplay setting is turned on. The sole reason they did that hack is to increase the framerate of the sensors when the main camera is not rendering.

Supaple-x, 2019-02-15:
> I realized I was using the latest version of the Unreal Engine (4.21) after installing using the default command:
> git clone https://github.com/EpicGames/UnrealEngine.git
> Instead, using the [...]

It can be executed in VS command line?

msb336, 2019-05-13:
Looks like this issue was fixed in [this pr](https://github.com/microsoft/AirSim/pull/1888). If you are still having a problem related to this thread, please reopen!

closed 2019-05-13 by msb336