microsoft/AirSim #1658: UE4.18 failed to start on ubuntu16.04

fangchuan, 2018-12-23:
Hi,  I meet a hard problem recently when i use AirSim in my  GPU server. I build the UE4.18 and AirSim successfully, but I couldn't start UE4Editor, because I have installed the nvidia driver in the server( i always get in stuck in a login loop if  ), the OpenGL version in my server is 3.0, su i need to start the UE4Editor like this:
`./path/to/UE4Editor -opengl3`
then the log file tell me the start up  program was crashed...
`[2018.12.23-08.11.33:596][  0]LogMaterial: Verifying Global Shaders for GLSL_150
[2018.12.23-08.11.33:607][  0]LogSlate: Using Freetype 2.6.0
[2018.12.23-08.11.33:609][  0]LogSlate: SlateFontCache - WITH_FREETYPE: 1, WITH_HARFBUZZ: 0
[2018.12.23-08.11.33:609][  0]LogSlate: SlateFontCache - WITH_FREETYPE: 1, WITH_HARFBUZZ: 0
[2018.12.23-08.11.34:240][  0]LogLinux: Selected Device Profile: [Linux]
[2018.12.23-08.11.34:532][  0]LogMeshReduction: Using QuadricMeshReduction for automatic static mesh reduction
[2018.12.23-08.11.34:532][  0]LogMeshReduction: No automatic skeletal mesh reduction module available
[2018.12.23-08.11.34:532][  0]LogMeshReduction: No automatic mesh merging module available
[2018.12.23-08.11.34:532][  0]LogMeshReduction: No distributed automatic mesh merging module available
[2018.12.23-08.11.34:532][  0]LogMeshMerging: No automatic mesh merging module available
[2018.12.23-08.11.34:532][  0]LogMeshMerging: No distributed automatic mesh merging module available
[2018.12.23-08.11.34:732][  0]LogNetVersion:  , NetCL: 0, EngineNetVer: 2, GameNetVer: 0 (Checksum: 3657919204)
[2018.12.23-08.11.34:993][  0]LogAssetRegistry: FAssetRegistry took 0.0103 seconds to start up
[2018.12.23-08.11.36:157][  0]LogPackageLocalizationCache: Processed 7 localized package path(s) for 2 prioritized culture(s) in 0.024712 seconds
[2018.12.23-08.11.36:164][  0]LogUObjectArray: 35531 objects as part of root set at end of initial load.
[2018.12.23-08.11.36:164][  0]LogUObjectAllocator: 7280144 out of 0 bytes used by permanent object pool.
[2018.12.23-08.11.36:164][  0]LogUObjectArray: CloseDisregardForGC: 0/0 objects in disregard for GC pool
[2018.12.23-08.11.36:399][  0]LogHAL: Linux SourceCodeAccessSettings: NullSourceCodeAccessor
[2018.12.23-08.11.37:405][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2018.12.23-08.11.37:405][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2018.12.23-08.11.37:405][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:405][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:406][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:406][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:406][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2018.12.23-08.11.37:406][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2018.12.23-08.11.37:407][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:407][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:407][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:407][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:407][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2018.12.23-08.11.37:407][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2018.12.23-08.11.37:407][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:407][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:408][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:408][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:408][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2018.12.23-08.11.37:408][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2018.12.23-08.11.37:408][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:408][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:409][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2018.12.23-08.11.37:409][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2018.12.23-08.11.37:415][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png' error.
[2018.12.23-08.11.37:415][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png
[2018.12.23-08.11.37:415][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2018.12.23-08.11.37:415][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2018.12.23-08.11.37:415][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2018.12.23-08.11.37:415][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2018.12.23-08.11.37:416][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png' error.
[2018.12.23-08.11.37:416][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png
[2018.12.23-08.11.37:416][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2018.12.23-08.11.37:416][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2018.12.23-08.11.37:416][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2018.12.23-08.11.37:416][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2018.12.23-08.11.37:417][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_ATC_24x.png' error.
[2018.12.23-08.11.37:417][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_ATC_24x.png
[2018.12.23-08.11.37:417][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2018.12.23-08.11.37:417][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[... cut ...]
              GPU:  8 * NVIDIA V100
              Virtualization: openstack
             Virtual Machine: 
                                 Ubuntu16.04, kernel version 4.15;
                                 nvidia driver version: 384.145(no opengl files);
                                 python: 3.6.5
                                AirSim: master branch
                               Unreal Engine:  4.18

[image]
[image]
[image]
[UE4EditorLog.txt](https://github.com/Microsoft/AirSim/files/2705874/UE4EditorLog.txt)

madratman, 2019-01-08:
there seem to be two issues in the really long debug message you posted. 
- libcurl and the whole slate png repetition, neither of which seem to me to be the actual cause, at least on the onset. 
perhaps try googling if your opengl version is compatible with your nvidia drivia version 
i'd also try posting or looking for similar bugs on https://answers.unrealengine.com

fangchuan, 2019-01-10:
> there seem to be two issues in the really long debug message you posted - libcurl and the whole png not found (repeated multiple times), neither of which seem to me to be the actual cause, at least on the onset.
> perhaps try googling if your opengl version is compatible with your nvidia drivia version
> i'd also try posting or looking for similar bugs on https://answers.unrealengine.com

> there seem to be two issues in the really long debug message you posted - libcurl and the whole png not found (repeated multiple times), neither of which seem to me to be the actual cause, at least on the onset.
> perhaps try googling if your opengl version is compatible with your nvidia drivia version
> i'd also try posting or looking for similar bugs on https://answers.unrealengine.com

i have tested the opengl 3 on the server, and it's ok. do you mean the problem is about libcurl?  i noticed libcurl want do some http connections but failed in the proxy, is it caused by my server cannot get access to google? Because the server is delegated in our datacenter of the university.

fangchuan, 2019-01-10:
@madratman 
i have tested the opengl 3 on the server, and it's ok. do you mean the problem is about libcurl? i noticed libcurl want do some http connections but failed in the proxy, is it caused by my server cannot get access to google? Because the server is delegated in our datacenter of the university.

jonyMarino, 2020-08-17:
closed due to age and not an airsim issue

closed 2020-08-17 by jonyMarino