microsoft/AirSim #1659: DllNotFoundException error for unity airsim version

yumianhuli2, 2018-12-23:
Follow the airsim for linux build , the error is below!
And I can not control the car by using WASD or arrow.

DllNotFoundException: AirsimWrapper
AirSimUnity.VehicleCompanion.InitDelegators () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:85)
AirSimUnity.VehicleCompanion..cctor () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:30)
Rethrow as TypeInitializationException: The type initializer for 'AirSimUnity.VehicleCompanion' threw an exception.
AirSimUnity.Vehicle.Start () (at Assets/AirSimAssets/Scripts/Vehicles/Vehicle.cs:58)
AirSimUnity.Car.Start () (at Assets/AirSimAssets/Scripts/Vehicles/Car/Car.cs:25)

mustfkeskin, 2019-01-04:
I have same problem at Windows10.

hardbroasd, 2019-04-07:
No DLL files are generated during the build process，leave you e-mail and I send you the file

msb336, 2019-04-08:
If you are using Ubuntu, the build script should generate a shared library (.so), not a dll. To see if it the library was built, navigate to `Unity/linux-build`. There should be a file in there called libAirSimWrapper.so . If that file doesn't exist, I would recommend cleaning out your repository and re-running the `Unity/build.sh` command.

Liukangjun01, 2019-05-19:
> 在构建过程中没有生成DLL文件，请留下电子邮件，然后我将文件发送给您

Hello, I also met the same problem, if you can please send me a DLL. liukangjun0000@outlook.com

Shwethakrishnaa, 2019-06-27:
Follow the airsim for windows build , the error is below!
And I can not control the car by using WASD or arrow, the car is not moving

DllNotFoundException: AirsimWrapper
AirSimUnity.VehicleCompanion.InitDelegators () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:85)
AirSimUnity.VehicleCompanion..cctor () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:30)
Rethrow as TypeInitializationException: The type initializer for 'AirSimUnity.VehicleCompanion' threw an exception.
AirSimUnity.Vehicle.Start () (at Assets/AirSimAssets/Scripts/Vehicles/Vehicle.cs:58)

windows 10. 
[image]

Denisa1997, 2020-02-20:
I have the same problem, can you please send me the DLL files
denisa_maria1997@yahoo.com

khizar-anjum, 2020-06-25:
@msb336 , having the same problem, maybe you can upload it on a drive and share it for everybody having the same problem.

khizar-anjum, 2020-06-27:
Turns out that only the latest bleeding edge of the Airsim API for Unity can be built, anything older has bugs. It is expected as Unity with Airsim is still experimental.

closed 2020-09-04 by jonyMarino

sanjaypAtel09, 2024-11-07:
@hardbroasd I have the same problem, can you please send me the DLL files.
sanjay.bhoy.if@gmail.com