microsoft/AirSim #1668: How to use the pre-compiled binaries for setting up the city environment with airsim using that released for linux? Can someone please explain how to use the pre-compiled binaries for the city environment using linux machine.

UmaArunachalam8, 2019-01-01:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

ssokhey, 2019-01-23:
@UmaArunachalam8 Just download the latest binaries (updated 11 days ago) and follow the updated instructions.

[https://github.com/Microsoft/AirSim/releases](url)

madratman, 2019-01-23:
Hey. the city environment is corrupt yes. I'll try to update it by EOD.

kylelindgren, 2019-02-04:
@madratman have you had a chance to fix the city environment binaries?

viniciusguigo, 2019-02-08:
I'm also interested and could not extract the files. Thank you!

MrDadaGuy, 2019-02-25:
$ 7z x CityEnviron.zip.001

7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=en_US.UTF-8,Utf16=on,HugeFiles=on,64 bits,48 CPUs Intel(R) Xeon(R) CPU E5-2687W v4 @ 3.00GHz (406F1),ASM,AES-NI)

Scanning the drive for archives:
1 file, 1782579200 bytes (1700 MiB)

Extracting archive: CityEnviron.zip.001
           
CityEnviron.zip
ERRORS:
Headers Error
Unconfirmed start of archive

CityEnviron.zip
WARNINGS:
There are data after the end of archive

--
Path = CityEnviron.zip.001
Type = Split
Physical Size = 1782579200
Volumes = 2
Total Physical Size = 3117962657
----
Path = CityEnviron.zip
Size = 3117962657
--
Path = CityEnviron.zip
Type = zip
ERRORS:
Headers Error
Unconfirmed start of archive
WARNINGS:
There are data after the end of archive
Physical Size = 3117932051
Tail Size = 30606

    
Would you like to replace the existing file:
  Path:     ./CityEnviron.sh
  Size:     261 bytes (1 KiB)
  Modified: 2018-08-15 14:34:38
with the file from archive:
  Path:     CityEnviron.sh
  Size:     261 bytes (1 KiB)
  Modified: 2018-08-15 14:34:38
? (Y)es / (N)o / (A)lways / (S)kip all / A(u)to rename all / (Q)uit? A

ERROR: Data Error : CityEnviron/Content/Paks/CityEnviron-LinuxNoEditor.pak
                                                               
Sub items Errors: 1

Archives with Errors: 1

Warnings: 1

Open Errors: 1

Sub items Errors: 1

zhangzhen886, 2019-03-20:
I also have same problem...

luxiangnk, 2019-03-26:
I got the same problem.. Thanks!

yerzhik, 2019-03-28:
Same problem. Hopefully can be fixed soon...

sebLopezCot, 2019-07-15:
Same issues.

gonzaloaguilarjimenez, 2019-08-21:
Same problem

bozcani, 2019-10-22:
Same problem. Still unsolved

jonyMarino, 2020-09-02:
Treated in #2407

closed 2020-09-02 by jonyMarino

JohnVorwald, 2021-12-16:
I don't see the solution in #2407.