microsoft/AirSim #1669: Poor performance

coldwarrior5, 2019-01-02:
I tried commenting inside similar issues but since they are closed and I am not getting any response I created a new issue, sorry for that.

I wanted to point out that the performance is abysmal the moment you start using sensors, and if you have a slightly detailed scene the performance drastically drops to well below 30 fps.
Also, I wanted to point out to a similar simulation tool, DeepDrive simulator, that uses shared memory to handle a similar problem.

> Our frame rates are higher while capturing several cameras, due to the use of shared memory rather than sockets and asynchronous transfer.

More information available at: https://deepdrive.io/blog/building-on-unreal.html
Source code available at: https://github.com/deepdrive/deepdrive

RVBldr, 2019-01-08:
Thanks Cold. We're aware of that and looking into performance improvements.

coldwarrior5, 2019-01-09:
Perfect, thank you so much.
I was myself looking how to improve this in the case you decided not to.
Carry on with amazing work.

petergerten, 2019-05-16:
I am curious: is someone already working on a shared memory implementation?

petergerten, 2019-07-09:
I really like AirSim but the performance is unusable for many things. I run it on an expensive V100 (p3.2xlarge on AWS) instance but it is still too slow. There is literally nothing money can buy that would improve the performance.

The shared memory approach sounds promising.

RVBldr, 2019-07-09:
Peter, can you provide some details (i.e., scenario, what sensors, etc)

jonyMarino, 2020-09-02:
#2472