microsoft/AirSim #1672: Using lidar to estimate collision

Ahmad5112, 2019-01-03:
I need to use LiDAR  to estimate collision for Quadcopter.
I am using block environment , Can anybody please guide me how I can estimate collision using lidar sensor and run collision avoidance algorithm accordingly

madratman, 2019-01-08:
I'd try to build a map with the lidar observations first. Have a look at [LOAM](https://github.com/laboshinl/loam_velodyne) and [hdl_graph_slam ](https://github.com/koide3/hdl_graph_slam) to begin with

Ahmad5112, 2019-01-09:
you recommend me to start with lidar mapping but I don't want to map the information. As lidar in AirSim returns the location of the point being hit by the lidar. I just Want to calculate the distances from the point and base in that distances I want to estimate collision Would it be the right approach to try .plese let me know as I don't want to go through a complex detection algo as it is not the part of my work

madratman, 2019-02-19:
Assuming you question is `I just Want to calculate the distances from the point and ... `, you can calculate the euclidean distance between two 3D points as sqrt( (x1-x2)^2 + (y1-y2)^2 + (z1-z2)^2). if you're lazy to do that, you can call [np.linalg.norm](https://docs.scipy.org/doc/numpy-1.13.0/reference/generated/numpy.linalg.norm.html)

`...and base in that distances I want to estimate collision` : well, here you should first understand and define what do you mean by "estimating collision". A popular metric you can use is `time to collision`. you can look at this [paper](https://www.sbes.vt.edu/gabler/publications/Kusano-Gabler-SAE-TTC_EDRs-2011-01-0576.pdf for details

collision is a binary concept, however you can easily come up with another continuous metric based on distance to nearest obstacle in your lidar readings. so you can do find the minimum reading. 
let's define `x` is this min reading. 
then you can cook up a "collision metric" (or whatever you mean by "estimating collision") by defining a minimum (`x_min`) and a maximum values (`x_max`) and define collision as "0" if the minimum lidar reading, `x=x_min` and "1" if `x=x_max`

this is essentially the [potential field](https://www.google.com/?q=potential+field+path+planning) idea

closed 2019-02-19 by madratman