microsoft/AirSim #1675: Unreal Editor crashes when Python script finishes before drone finishes moving

Spiraseven, 2019-01-03:
I am on Ubuntu 16.04

My python script only contains a take off command and client.moveToPositionAsync(-10, 10, -5, 1)

I am not using .join()

So after the moveToPosition command the script ends, which should cause the client connection to close as well. However, this causes the Unreal Editor to crash shortly after. Something about the script ending and the client closing while the drone is still moving to the position I think causes the crash. Is there a way around this, or a best practice? Is the only option to have a loop that checks status and waits or cancels the last action if I want to end the python script?

clovett, 2019-01-08:
either way python scripts should not be able to crash AirSIm, this is a bad bug that needs to get fixed.  I run into it all the time when I just hit "ctrl+c" to interrupt a script.  Seems to be a bug in rpclib.

Spiraseven, 2019-01-08:
I was testing a case where the drone never makes it to the destination due to something blocking it. So I was just letting the script end instead of staying hung up from join(). There are good ways to go about this, such as checking for collisions and stopping the move command then ending the script. In my final script I will probably not run into this issue. But during development and testing there are times like clovett said where a ctrl+c or letting the script end is just easier for python work. It be nice if unreal didn't crash so I don't have to restart it, just adds dev time.

clovett, 2019-01-08:
I think if you finish the script with a client.hoverAsync().join() it will flip it out of "offboard" mode and script termination will be safer.

closed 2020-09-04 by jonyMarino