microsoft/AirSim #1677: Incorrect behavior of MoveToPositionAsync #1643，how to solve it?

love-mj, 2019-01-04:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
[](url)
[image]

majouda7, 2019-01-08:
I have experienced exactly the same issue (I posted it in Incorrect behavior of MoveToPositionAsync #1643). It seems that AirSim is not aware about this offset in unreal because when I call client.simGetVehiclePose().position or client.simGetGroundTruthKinematics().position, it returns exactly the same requested position.
My setting.json file looks like this:

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "AutoCreate": true,

      "Sensors": {
        "LidarSensor1": {
          "SensorType": 6,
          "Enabled": true,
          "Range": 100,
          "NumberOfChannels": 2,
          "RotationsPerSecond": 10,
          "PointsPerSecond": 100000,
          "X": 0,
          "Y": 0,
          "Z": -0.8,
          "Roll": 0,
          "Pitch": 0,
          "Yaw": 0,
          "VerticalFOVUpper": 0,
          "VerticalFOVLower": 0,
          "HorizontalFOVStart": 0,
          "HorizontalFOVEnd": 359,
          "DrawDebugPoints": true,
          "DataFrame": "SensorLocalFrame"
        }
      }
    }
  }
}

love-mj, 2019-01-09:
@lovettchris My setting.json is the default mode,
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
		"PX4": {
		   "VehicleType": "PX4Multirotor",
		   "SerialPort": "*"
		}
	}
}

saihv, 2019-01-15:
I think I have the same problem. I am trying to control multiple SimpleFlight drones with the Python API, and I am encountering errors in how MoveToPositionAsync actually moves the drones. For instance, this is a set of commands I gave the drone:

```
f1 = client.moveToPositionAsync(5, 0, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(-5, 0, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(-5, -10, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(5, -10, -8, 3, vehicle_name="Drone1")
f1 = client.moveToPositionAsync(5, 0, -8, 3, vehicle_name="Drone1")
```

My world is properly aligned, so this should make the drone fly in a perfect square. Instead, this is what the drone does (plotted from the airsim_rec log):

[image]

Unless I am doing/understanding something wrong, this seems to be a pretty major issue.

jasonhbartlett, 2019-01-16:
@lovettchris  No, sir.  The script you posted.  I cannot get 2 moveToPositionAsync().Join() calls to ever work.  Certainly not 3 or more.  Even while putting extra client.enableApiControl(True) all over the place.  No dice, I lose control to the failsafe.  I'm guessing the problem is with the PX4 params.  Could you or someone provide, or could ya'll add to the AirSim repo an generic PX4 params test file that could be used by a new user to help us verify our PX4 sitl setup?  params that are optimized to work with all of the test Api's like HelloDrone, etc?  I posted my param here: [show params](https://gist.github.com/jasonhbartlett/d728aecf32a7a091d93df2d4eee70349)

jasonhbartlett, 2019-01-16:
@lovettchris 
By the way, this is what that same script looks like with SimpleFlight.  It's actually damn beautiful.  I need to get these same (or at least very similar) results with my PX4 sitl setup.
[image]
[image]

majouda7, 2019-01-16:
@jasonhbartlett Did you check if the position of the drone in unreal is actually as requested? In my case, using SimpleFlight, the drone overshoots the requested position as described in #1643. This behavior varies considerably with speed.

jasonhbartlett, 2019-01-16:
Hi @majouda7 
Yep, I'm getting the same results:
Running in Unity 5m/s :
[image]

Running in Unreal 5m/s :  (Although it seemed to correct itself over time in Unreal)
[image]

I can chart the path in Unity.
[image]
I don't know how to show the trail to chart the path in Unreal.  :(

majouda7, 2019-01-16:
@jasonhbartlett, in my case the drone doesn't seem to correct itself in unreal over time. This what I get if I put a sleep after each move. It is obvious that the drone is drifting over x and y axes. 
[image]

lovettchris, 2019-01-16:
@jasonhbartlett regarding parameters, I did put some info here [moveOnPath demo](https://github.com/Microsoft/AirSim/wiki/moveOnPath-demo).  Saving all parameters would be a maintenance problem because px4 might add or change other unrelated parameters.  But the ones I usually use are:
```
param set SYS_MC_EST_GROUP 2
param set MPC_XY_VEL_MAX 20
param set MPC_XY_CRUISE 5
param set COM_OBL_ACT 1
param set COM_OBL_RC_ACT 5
param set NAV_RCL_ACT 0
param set NAV_DLL_ACT 0
```
There's also some in [px4_sitl](https://github.com/Microsoft/AirSim/blob/master/docs/px4_sitl.md).  I have found a bug in airsim to do with [magnetic declination](https://github.com/Microsoft/AirSim/issues/1340), so until that is fixed I recommend flying with latitude 0.  

I put my full set of [params here](https://gist.github.com/lovettchris/c40eb813d5186fe5d1fdf1ce221fbae8).  

Let's also rule out C++ versus python, can you run this python test script and paste the results (as text) ?  Thanks.

```python
import sys
import time

import setup_path 
import airsim

def print_position(client):    
    state = client.getMultirotorState()
    gps = state.gps_location
    print("gps location lat={}, lon={}, alt={}".format(gps.latitude, gps.longitude, gps.altitude))
    pos = state.kinematics_estimated.position
    print("local position x={}, y={}, z={}".format(pos.x_val, pos.y_val, pos.z_val))

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

print_position(client)

landed = client.getMultirotorState().landed_state
if landed == airsim.LandedState.Landed:
    print("taking off...")
    client.takeoffAsync().join()
else:
    client.hoverAsync().join()

if client.getMultirotorState().landed_state == airsim.LandedState.Landed:
    print("take off failed, please check message log")
    sys.exit(1)

# AirSim uses NED coordinates so negative axis is up.
# z of -7 is 7 meters above the original launch point.
z = -7

# make sure we are at the start location (previous flight might have missed the landing spot by a bit.)
client.moveToPositionAsync(0,0,z,1).join()

print("flying path...")

# this method is async, but we are calling .join() so the script waits for path to complete.
for i in range(100):
    time.sleep(3)      
    pos = client.simGetGroundTruthKinematics().position
    print("{:.3f},{:.3f},{:.3f}".format(pos.x_val, pos.y_val, pos.z_val))
    client.enableApiControl(True)
    client.moveToPositionAsync(-5,0,z,1).join()
    pos = client.simGetGroundTruthKinematics().position
    print("{:.3f},{:.3f},{:.3f}".format(pos.x_val, pos.y_val, pos.z_val))
    time.sleep(3)
    pos = client.simGetGroundTruthKinematics().position
    print("{:.3f},{:.3f},{:.3f}".format(pos.x_val, pos.y_val, pos.z_val))
    client.enableApiControl(True)
    client.moveToPositionAsync(5,0,z,1).join()
    pos = client.simGetGroundTruthKinematics().position
    print("{:.3f},{:.3f},{:.3f}".format(pos.x_val, pos.y_val, pos.z_val))

client.moveToPositionAsync(0,0,-10,1).join()
print("landing")
client.landAsync().join()
print("disarming drone")
client.armDisarm(False)
client.enableApiControl(False)
print("done!")

print_position(client)
```
With this test script I am seeing some sort of AirSim bug where sometimes the moveToPositionAsync call terminates prematurely, possibly thinking it has already reached the intended destination, when it clearly has not.  So my graph below shows this problem - it should be a sharp tooth diagram back and forth like you see it do a few times, but most of the time it gets stuck on one end or the other for a while:

[image]

So this will need some airsim debugging...  But I did not see any "Accel #0 fail" or "Gyro #0 fail" errors.  I do see lots of "Failsafe enabled:" warnings, which you can safely ignore.

majouda7, 2019-01-16:
@lovettchris, Do you have any insights on the variation (offset) of the drone's arriving position with respect to its velocity ? Is there any plans to fix that bug?

jasonhbartlett, 2019-01-17:
@lovettchris 
I'm almost certain I'm seeing that bug you described where the moveToPositionAsync call terminates prematurely.  Indeed, (after making sure the default state was changed to "Hold" for all failsafe triggers so that the drone doesn't try to "Land" or "RTH") **I am seeing that exact pattern.**  Also getting lots of "Failsafe enabled:" warning. The drone flies back and forth but sometimes get stuck on one side or the other for a bit.  After awhile, one of the moveToPositionAsync calls must get through and it is able to move across to the other side again.
Here are the results of running your python script:
```dos
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

gps location lat=47.6414680480957, lon=-122.1401596069336, alt=120.14900207519531
local position x=0.0, y=0.0, z=0.0
taking off...
flying path...
-0.056,-0.006,-4.939
-0.055,-0.007,-4.938
-0.226,-0.066,-5.019
-0.212,-0.073,-5.022
0.076,0.028,-5.088
-5.208,-0.109,-5.065
-4.943,-0.588,-5.054
5.077,0.023,-5.064
4.994,0.076,-5.136
-5.027,-0.068,-5.076
-4.965,-0.145,-5.103
5.090,0.064,-5.077
4.982,0.108,-5.141
1.654,0.020,-5.087
1.318,-0.527,-5.063
5.254,0.033,-5.064
5.084,0.562,-5.045
-5.044,-0.052,-5.056
-4.962,-0.093,-5.045
5.111,0.068,-5.079
4.972,0.113,-5.048
-1.006,-0.228,-5.101
-1.241,-0.614,-5.105
5.139,0.373,-5.074
4.880,0.552,-5.084
-5.121,-0.044,-5.044
-4.990,-0.094,-5.066
5.112,0.057,-5.047
5.003,0.111,-5.047
5.001,0.107,-5.047
4.844,-0.170,-5.055
5.064,-0.299,-5.101
5.066,-0.250,-5.115
5.065,-0.248,-5.116
4.847,-0.283,-5.104
4.947,-0.348,-5.075
4.989,-0.318,-5.040
4.989,-0.316,-5.040
4.773,-0.333,-5.024
4.900,-0.407,-5.081
4.900,-0.357,-5.079
-5.084,-0.113,-5.075
-4.917,-0.161,-5.110
5.120,0.050,-5.070
4.983,0.107,-5.102
0.731,-0.048,-5.016
0.413,-0.572,-5.041
5.207,0.177,-5.077
4.976,0.622,-5.029
-5.056,-0.059,-5.053
-4.965,-0.111,-5.054
5.092,0.065,-5.073
4.991,0.126,-5.104
-5.122,-0.036,-5.031
-5.005,-0.087,-5.099
5.114,0.050,-5.009
4.986,0.111,-5.025
-5.092,-0.052,-5.097
-4.964,-0.107,-5.078
[... 328 lines cut ...]
-4.633,0.019,-5.031
-4.431,0.016,-5.056
-4.746,0.058,-5.073
-4.958,0.013,-5.121
-4.959,0.012,-5.120
-4.639,-0.006,-5.088
-4.637,-0.004,-5.088
-4.567,0.100,-5.064
-4.571,0.101,-5.064
-4.508,0.076,-5.057
-4.707,0.050,-5.050
-4.757,0.007,-5.002
5.105,0.189,-5.136
landing
disarming drone
done!
gps location lat=47.641510009765625, lon=-122.14017486572266, alt=119.40699768066406
local position x=0.0, y=0.0, z=0.0
Press any key to continue . . .
```
And here's a picture showing the complete flight path.  (I'll have to start publishing these as abstract art :)
[image]
BTW - Although indeed the starting local position was 0, 0, 0, as you can see the ending position was not, therefore I don't know why the final check of local position reported 0, 0, 0.  The LAT/LON is probably accurate.

Please let me know what else I can do to help test things out.  Hopefully this bug is traceable.  :) Thanks sir!

lovettchris, 2019-01-17:
Ok, great, so we have a repro for the bug and we should make this a high priority item.  I also see the final print_position 0,0,0 but which is another bug, not sure what's happening there, perhaps changing the armed state or the enableApiControl state is impacting the position printed (which it shouldn't)...

whxru, 2019-03-06:
I met the same problem and unfortunately I met another problem #1292 too. Finally I found that I could use `moveByVelocityAsync(vx, vy, vz, duration)` to set velocity to make the drone move and use `moveByVelocityAsync(0, 0, 0, duration)` to make it stop after a specific time . In this way I could custom a new `moveToPosition` method like:
```python
def moveToPosition(x, y, z, v):
    currentPos = uav.getMultirotorState().kinematics_estimated.position
    t = ((currentPos.x_val - x)**2 + (currentPos.y_val - y)**2 + (currentPos.z_val - z)**2)**0.5 / v
    delta_x = x - currentPos.x_val
    delta_y = y - currentPos.y_val
    delta_z = z - currentPos.z_val
    vx = delta_x / t
    vy = delta_y / t
    vz = delta_z / t
    uav.moveByVelocityAsync(vx, vy, vz, t)
    time.sleep(t)
    uav.moveByVelocityAsync(0, 0, 0, 5)
    uav.hoverAsync().join()
```
This method is able to make the drone stop at the given position accurately even with a fast speed (e.g. 10m/s in my simple test). So this might be a temporal solution before the bug has been fixed.

seheevic, 2019-11-14:
I've encountered this problem and investigated a little inside AirSim source code.
There are 2 problems I can state here with respect to 'termination criteria' of moveOnPath.

1. MultirotorApiBase::moveOnPath
https://github.com/microsoft/AirSim/blob/1b10a49b004bf56195897813b6d3b4394b906f88/AirLib/src/vehicles/multirotor/api/MultirotorApiBase.cpp#L202-L208
When **velocity** argument which determines lookahead (default behavior of adaptive_lookahead) is too high compared to the path length, **next_path_loc** will be set to the very last path segment, so the code inside loop is never executed since initial goal_dist was set to zero. It seems that we need more robust 'termination criteria'.

2. Physics frequency / Command frequency / Rendering frequency when using Multirotor
[Default physics frequency: 333 Hz (3ms)]
https://github.com/microsoft/AirSim/blob/1b10a49b004bf56195897813b6d3b4394b906f88/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldBase.h#L65
[Default command frequency: 50 Hz (20ms)]
https://github.com/microsoft/AirSim/blob/1b10a49b004bf56195897813b6d3b4394b906f88/AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightApi.hpp#L181-L184
[Rendering frequency: not fixed, but let's say it's about 30 Hz (a little low frequency)]
AirSim(SimpleFlight) updates **estimated kinematics information** (which will be used in the command loop) from high frequency physics results on main thread every tick, so the update frequency is as same as rendering frequency. But as you can see from source code, when the command frequency is higher than the rendering frequency, I think it would be possible that control command is processed twice within single main thread tick. That can also cause **goal_dist** to be zero (no movement from measurement), and the command loop will be terminated unexpectedly.

3. detectTakingOff which affects control(goal) time out detection of OffboardApi
This seems to be fixed or at least improved.

I'm not sure that this is the reason of the discussion above.
But it seems to be somewhat related to them.

xxEoD2242, 2020-03-28:
So, this may not solve anyone's problems, but I had a lot of issues in using the join calls for calling absolute positioning and I what I have determined is that when you using loops inside of Python to do that type of async behavior, either Unreal, Airsim or RPC freaks out about the asynchronous method.

To solve my issues, as I was writing a simple chase-follow script, I did the following. This isn't refactored in any way but I haven't had any issues with positioning since.

```
for i, waypoint in enumerate(route.primary_route):
    try:
        next_point = route.primary_route[i + 1]
    except Exception as error:
        print('Final point!')
        next_point = False
    f1 = client.moveToPositionAsync(waypoint[0], waypoint[1], waypoint[2], waypoint[3], vehicle_name=veh_1_name)
    if (i % 10 == 0):
        f2 = client.moveToPositionAsync(waypoint[0], waypoint[1], waypoint[2] - 0.1, waypoint[3] + 1, vehicle_name=veh_2_name)
    else:
        f2 = client.moveToPositionAsync(waypoint[0], waypoint[1], waypoint[2], waypoint[3], vehicle_name=veh_2_name)

    if (i % 5 == 0):
        print(waypoint)

    # Instead of join, pull state and get GPS data to confirm where the drone is at
    kinematics = client.simGetGroundTruthKinematics(vehicle_name=veh_1_name)
    if next_point:
        # For the first point, the drones need to fly to a normalized location
        if i == 0:
            f2.join()
        else:
            while (abs(kinematics.position.x_val - abs(next_point[0])) - 3) > CONFIRMATION_DISTANCE and (abs(kinematics.position.y_val - abs(next_point[1])) - 3) > CONFIRMATION_DISTANCE and abs(abs(kinematics.position.z_val) - abs(next_point[2])) > CONFIRMATION_DISTANCE:
                kinematics = client.simGetGroundTruthKinematics(vehicle_name=veh_1_name)
                print('\n')
                print(kinematics.position.x_val)
                print(kinematics.position.y_val)
                print(kinematics.position.z_val)
                print((abs(kinematics.position.x_val - abs(next_point[0])) - 3))
                print((abs(kinematics.position.y_val - abs(next_point[0])) - 3))
                print(abs(abs(kinematics.position.z_val) - abs(next_point[2])))
                print('\n')
                time.sleep(0.2)
```

cloudlakecho, 2020-05-21:
@xxEoD2242, Thanks for the snippet. 
I wonder how to choose the sleeping time. Is it empirical depending on position distance?

cloudlakecho, 2020-05-22:
@saihv, would you explain how to plot the vehicle trajectory using ` airsim_rec log`? I wonder your coordinate is rotated. Have you tried to see your drone GPS location as well?

cloudlakecho, 2020-05-27:
@lovettchris, thanks for posting nice graph of drone location. I try to plot, but I couldn't find how to get the position continuously. Please, share the function of recording drone location, thanks.

lovettchris, 2020-05-28:
I was using PX4 drone with a mavlink logger, probably connected to the LogViewer app included in this repo.

PR #3545 (2021-04-15): Update moveToPosition to use setpoint control message instead of velocity control message

Fixes: #1677

## About
Override moveToPosition() implementation in MavLinkMultirotorApi.hpp to use MavLink setpoint control messages for movement instead of velocity control messages. In place of velocity control messages, the MPC_XY_CRUISE parameter is set to the inputted velocity for the duration of the moveToPosition() call and then reset to its original value at the end of the call.

## How Has This Been Tested?
tested with hello_drone3.py script in description of #1677 

## Screenshots (if appropria [... cut]
closed 2021-04-15 by jonyMarino