microsoft/AirSim #1678: Unity Linux build error

fawadahm, 2019-01-06:
I tried to build AirSim for Unity on Linux. I was able to install Unity successfully but faced the following errors when I tried to run setup.sh

```
./setup.sh 
+ set -e
+++ dirname ./setup.sh
++ cd .
++ pwd
+ SCRIPT_DIR=/media/fawad/DATA/AirSim
+ pushd /media/fawad/DATA/AirSim
+ downloadHighPolySuv=true
+ [[ '' == \-\-\n\o\-\f\u\l\l\-\p\o\l\y\-\c\a\r ]]
++ uname
+ '[' Linux == Darwin ']'
+ [[ ! -z '' ]]
+ sudo apt-get install -y build-essential
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 253 not upgraded.
++ lsb_release -rs
++ cut -d. -f1
+ VERSION=16
+ '[' 16 -lt 17 ']'
+ wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key
+ sudo apt-key add -
--2019-01-05 17:30:21--  http://apt.llvm.org/llvm-snapshot.gpg.key
Resolving apt.llvm.org (apt.llvm.org)... 151.101.26.49, 2a04:4e42:6::561
Connecting to apt.llvm.org (apt.llvm.org)|151.101.26.49|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3145 (3.1K) [application/octet-stream]
Saving to: ‘STDOUT’

-                       100%[===============================>]   3.07K  --.-KB/s    in 0.001s  

2019-01-05 17:30:21 (4.02 MB/s) - written to stdout [3145/3145]

OK
+ sudo apt-get update
Hit:1 http://apt.llvm.org/xenial llvm-toolchain-xenial-6.0 InRelease
Ign:2 http://dl.google.com/linux/chrome/deb stable InRelease                                   
Hit:3 https://nvidia.github.io/libnvidia-container/ubuntu16.04/amd64  InRelease                
Hit:4 http://us.archive.ubuntu.com/ubuntu xenial InRelease                                     
Hit:5 https://nvidia.github.io/nvidia-container-runtime/ubuntu16.04/amd64  InRelease           
Hit:6 https://repo.skype.com/deb stable InRelease                                              
Hit:7 https://nvidia.github.io/nvidia-docker/ubuntu16.04/amd64  InRelease                      
Hit:8 http://dl.google.com/linux/chrome/deb stable Release                                     
Get:9 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]                    
Get:10 http://security.ubuntu.com/ubuntu xenial-security InRelease [107 kB]                    
Hit:11 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu xenial InRelease                   
Hit:12 https://download.sublimetext.com apt/stable/ InRelease                                  
Ign:13 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial InRelease                   
Get:14 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease [107 kB]                 
Hit:16 http://ppa.launchpad.net/webupd8team/sublime-text-2/ubuntu xenial InRelease             
Ign:17 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial Release    
Ign:18 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 Packages
Ign:19 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main i386 Packages
Ign:20 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main all Packages
Ign:21 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en_US
Ign:22 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en
Ign:23 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:24 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:18 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 Packages
Ign:19 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main i386 Packages
Ign:20 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main all Packages
Ign:21 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en_US
Ign:22 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en
Ign:23 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:24 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:18 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 Packages
Ign:19 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main i386 Packages
Ign:20 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main all Packages           
Ign:21 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en_US      
Ign:22 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en         
Ign:23 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 DEP-11 Metadata  
Ign:24 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main DEP-11 64x64 Icons     
Ign:18 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 Packages         
Ign:19 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main i386 Packages          
Ign:20 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main all Packages           
Ign:21 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en_US      
Ign:22 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en         
Ign:23 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 DEP-11 Metadata  
Ign:24 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main DEP-11 64x64 Icons     
Ign:18 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 Packages         
Ign:19 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main i386 Packages          
Ign:20 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main all Packages           
Ign:21 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en_US      
Ign:22 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en         
Ign:23 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 DEP-11 Metadata  
Ign:24 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main DEP-11 64x64 Icons     
Err:18 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 Packages         
  404  Not Found
Ign:19 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main i386 Packages          
Ign:20 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main all Packages
Ign:21 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en_US
Ign:22 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main Translation-en
Ign:23 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:24 http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Fetched 323 kB in 14s (23.0 kB/s)
Reading package lists... Done
W: The repository 'http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/ubuntu-audio-dev/ppa/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

And then when I tried to run build.sh

```
The llvm-souce-50 folder was not found! Mystery indeed.
ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine
Please run setup.sh first.
```

madratman, 2019-01-08:
This is not an unity airsim bug, but an issue with your ppas. https://askubuntu.com/questions/65911/how-can-i-fix-a-404-error-when-using-a-ppa-or-updating-my-package-lists might help, else you should google the error messages

madratman, 2019-01-15:
Please re-open/comment if this is not resolved.

closed 2019-01-15 by madratman

ardianumam, 2019-04-09:
I get this exactly same error. Anyone already solve this?