microsoft/AirSim #1680: Low level control of Quadrotor

juanmed, 2019-01-07:
Hello! I would like to use Airsim for RL with the quadrotor. I am new to AirSim so I hope my question is not trivial nor that it has been already answered. 

Apart from RL, there is also low-level controller I would need to implement to use the same framework for the complete simulation of the system. As in #808, (closed) #[99](https://github.com/Microsoft/AirSim/issues/99), I would also like to work on low level control of the quadrotor. After reading on the previous issues, I got an idea of the current state of development for low level control of the quadrotor in Airsim. If I am not wrong, currently the lowest level of control is provided by moveByAngleThrottleAsync which allows to specify pitch and roll angles together with a throttle in order to change the orientation of the drone. This is not enough low-level control for me since I want to control the position and orientation through speed commands to the rotors and feedback from the sensors mounted in the drone.   I would like to ask the following questions:

1) Is there any documentation on simple_flight controller implementation? I understand using the provided APIs are not fast enough for (as real as possible) real-time control as mentioned in #99. So given the current state of airsim for low-level control I would like to base my controller on whatever simple_flight is doing. I understand it implements PD control laws, so I want to base my controller scripts on simple_flight in order to change the control law. 

2) I understand the focus on AirSim is RL. That being said, could anyone give me some pointers on how to approach the implementation of a controller on AirSim?  I know I need as a basis a way to get "raw" sensor data from the drone (in order to estimate my states) and a way to command rotor velocities to the drone, in such a way that the physics of the drone respond to the rotor velocities as in real life. Pointers on the implementation would be very welcome. (Of course the control law is my part)

Your comments will be very appreciated.

juanmed, 2019-01-17:
Regarding this functionality. In #808, user @zbenic was able to implement moveByRotorSpeed methods. He was kind enough to provide me help even he made this changes a long time ago (11months) and is not working with Airsim anymore. 

So, if one clones his MultirotorRL repository and make the changes in this 2 commits, it is possible to actually command rotor speeds (using the moveByRotorSpeed method) and have the simulated drone in UE4 move accordingly. It is however necessary to use the Block environment in his repository, since the as airsim plugin has changed since his changes.

The steps are as follows:
1) Clone his repo.
2) Make the following changes.
3) Open the blocks environment in his repo with UE4.
4) Write a simple script using the moveByRotorSpeed method.

There is one detail though: As far as I understand, It is not actually commanding rotor speeds per se, it actually converts the input rotor speed command to a torque. The effect of the force product of the rotor's propeller rotation is a torque around the center of mass so it makes sense. I would be nice however to command the rotor speeds themselves. 

I will try to port his changes to the current release and try to understand how is it possible to add api's for Unity. About adding APIs, there is an example simPrintLogMessage commit, but after looking at it, seems it is outdated.

juanmed, 2019-01-17:
Since I was not able to find an example commit for Unity in which an api is added, I will try to understand this [commit]( https://github.com/Microsoft/AirSim/pull/1424/commits/88d2ee6dc74e9ca274c63b5812f35972abca1951).

linxiaobo110, 2019-03-12:
@juanmed 
hello. I am doing simulation about quadrotor using airsim with python, which needs the low level control interface. Do you know how to get the interface in python? Can you help me ? My e-mail is linxiaobo110@gmail.com

linxiaobo110, 2019-03-12:
can we change the lastest version of airsim for using the low level control api

closed 2021-08-30 by jonyMarino