microsoft/AirSim #1682: rotateToYawAsync() does nothing

Spiraseven, 2019-01-08:
Hopefully I am just using this wrong, but when trying to rotate using rotateToYawAsync() nothing happens to the drone. I simply take off then call the rotate method.

client.rotateToYawAsync(90,5,1)

Has anyone had any luck using this. Is there a better way to have a drone turn in place? Is that possible?

gurpreetss, 2019-01-09:
Please try with client.rotateToYawAsync(90,5,1).join()

Spiraseven, 2019-01-09:
Yea nothing happens with adding join either. I am not seeing any errors. Maybe the function is just not hooked up? I have not gone through all the source code

gurpreetss, 2019-01-09:
I tried the function today and it worked, though there were some glitches.
Is it possible to share a code snippet?

Spiraseven, 2019-01-09:
I forgot to mention I was using the python api and I am on ubuntu as well working off of the master branch from December. Here is the code

`
import setup_path
import airsim
import numpy as np
import os
import tempfile
import pprint
import cv2
from PIL import Image
from array import array
import time

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

airsim.wait_key('Press any key to takeoff')
client.takeoffAsync().join()

airsim.wait_key('Press any key to rotate')
client.rotateToYawAsync(90,5,1).join()
`

chaitanyantr, 2019-01-17:
Yes...!for me too its same,

The 4th param of rotateToYawAsync() expecting vehicle name...don''t know what to fill

 **def rotateToYawAsync(self, yaw, timeout_sec = 3e+38, margin = 5, vehicle_name = ''):
        return self.client.call_async('rotateToYaw', yaw, timeout_sec, margin, vehicle_name)**

had given the command like.....

airsim.wait_key('Press any key to rotate')
client.rotateToYawAsync(90,5,1).join()

(Airsim version -  1.2)

whatseven, 2019-02-26:
I have the same problems. I test with `rotateByYawRateAsync`. It rotates. But when change it to `rotateToYawAsync` it fails and does nothing.

abhimanyuchadha96, 2019-07-10:
Has this been fixed?

alexander-hamme, 2019-08-15:
I am also experiencing this issue on Ubuntu 16. Has this been patched?

dimikout3, 2019-10-01:
I can use the rotateToYawAsync() but after a number of ratations the vehicles does not respond ...

PR #2516 (2020-04-08): Fix bug: rotateToYaw not working
Fix #1682, #856

Changes to fix `rotateToYaw`:
- `isYawWithinMargin` should receive the target yaw degree as the first argument
- The original stop criterion does not stop rotation due to inertia, so the resulted yaw is not guaranteed.

By the way, the `rotateByYawRate` implementation were made consistent with other API.
closed 2020-04-08 by madratman