microsoft/AirSim #1683: MultiRotor assets missing for Unity in AirSim

zsfVishnu, 2019-01-08:
The drone assets seem to be missing in the Unity assets folder for DroneDemo. The car works fine . If anyone knows how to add the drone assets into the simulation , do reply.

zsfVishnu, 2019-01-09:
@madratman , i followed the instructions mentioned in the doc and ran SimModeSelector only. It works fine for the car, but while choosing the drone it's not. There is no drone visible. I think it might have something to do with the assets within the Standard Assets folder [here](https://github.com/Microsoft/AirSim/tree/master/Unity/UnityDemo/Assets/Standard%20Assets/Vehicles) .The car model is present with all the texture and color support files etc , but there is none for Drone. I tried creating the MultiRotor folder and copied the fbx files for the drones but an error popped up. 
ps: I've also changed the json file i've switched over to drones.

juanmed, 2019-01-09:
If it helps to know: I installed Airsim for Unity for Ubuntu 16.04 some days ago and both the car and drone simulations appear and work as expected. I simple followed the instructions indicated by @madratman in usage.

juanmed, 2019-01-09:
@zsfVishnu There is one "Vehicles" folder and one "Vehicles.meta" file  in: AirSim/Unity/UnityDemo/Assets/Standard Assets, and inside that "Vehicles" folder there is on "Car" folder and one "Car.meta" file. There is no "Drone" folder inside the "Vehicles" nor the "Standard Assets" folder.

However, in AirSim/Unity/UnityDemo/Assets/AirSimAssets/Models  there are "UnityDrone.fbx" and "UnityDrone.fbx.meta" files. I have no idea what these files are and mention only for reference.

zsfVishnu, 2019-01-09:
yea, i'll just try to rebuild everything . I thought that the missing asset might be the issue.

zsfVishnu, 2019-01-09:
@juanmed , it didn't work for me. The cardemo unity file has the c# script for the car and all other dependencies but the dronedeom does not. You can see this in Unity on the left panel(drop dowm button) once you double click the SimModeSelector

msb336, 2019-01-15:
@zsfVishnu 
What version of unity are you using? Older versions of unity have different formatting for prefabs and in-game objects and are not compatible with newer prefabs.

To make sure you have the right version, follow the steps [here](https://github.com/Microsoft/AirSim/tree/master/Unity).

msb336, 2019-01-24:
Closing this thread due to inactivity.
@zsfVishnu if you are still having issues, please reopen.

closed 2019-01-24 by msb336