microsoft/AirSim #1690: Can't extract the CityEnviron.zip in win10

nighthb2019, 2019-01-10:
Hi , I've been downloaded the CityEnviron.zip.001 and the CityEnviron.zip.002 for ubuntu16.04 . And I  used 7zip to extract them, but still can't extract the CityEnviron.zip , it reportd errors. 
I decompressed in the WIN10 environment.
Does anyone know this reason?
Thanks.

madratman, 2019-01-15:
Just use the 7zip Gui and seleect the zip.001 file

closed 2019-01-15 by madratman

nighthb2019, 2019-01-20:
Hi,

I used the 7ZIP GUI to extract the 001.ZIP file, but I still get an error. What should I do?

madratman, 2019-01-23:
this was closed prematurely, we're working on updating the city binary soon

2016CS112, 2019-06-29:
@madratman , sir city binary updated or not?

2016CS112, 2019-06-29:
[image]

How to Extract this assets?