microsoft/AirSim #1691: How to Get the Signal Value of Px4 Handle Channel？

love-mj, 2019-01-11:
[image]
[image]
What function to get the button channel？Getting values in red boxes like simpleflight？

lovettchris, 2019-04-07:
See https://docs.px4.io/en/getting_started/rc_transmitter_receiver.html

love-mj, 2019-04-09:
[image]

GetRCChannal () is the access handle channel, but the value obtained is incorrect.

rebootVehicle () is to restart pixhawk, but pixhawk will be out of control after restart.
setupAirframePage () is a change in rack type, but it doesn't work.

love-mj, 2019-05-06:
获取通道信号已解决

closed 2019-05-06 by love-mj