microsoft/AirSim #1692: Weathering Effects on Airsim Neighborhood

sjnah, 2019-01-11:
Hello. 
I am in the process of extracting img / state data using Airsim.
I tried to put a weathering effect to lower the reliability of img. 

In the Blocks Environment, **simEnableWeather(True)** command worked normally. 
But it did not work in **Airsim Neighborhood**. 
The error codes are shown below.
-----------------------------------------------------------------------------------------------------------
_Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "test1.py", line 8, in <module>
    client.simEnableWeather(True)
  File "/usr/local/lib/python3.5/dist-packages/airsim/client.py", line 83, in simEnableWeather
    return self.client.call('simEnableWeather', enable)
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simEnableWeather' with argument count 1._
-----------------------------------------------------------------------------------------------------------

 I understand that the AirsimNH map is a copyrighted map and can not be separated by UnrealEngine. If you purchase the Modular Neighborhood Package in the market, can you use the weather effect normally?

Best Regards
S.J. Nah

RVBldr, 2019-01-15:
Weather effects such as rain and snow accumulation on surfaces has not been enabled in Neighborhood. With the latest AS code, you can see the rain/dust/fog/snow particle effects, but you won't see accumulations on surfaces without some changes to the materials.

sjnah, 2019-01-16:
Sorry. I think I chose the wrong words.
It does not mean that I want effects like snow accumulation.

The weather-related code or F10 key did not work on the AirSimNH map on the Release page.

Is the weather contents missing from the AirSimNH(Release version) map's Airsim Plugins?
If I use 'Modular Neighborhood' map, is it possible to use weather function?

sjnah, 2019-01-16:
Thank you a lot.

S.J.Nah
Best Regards

RVBldr, 2019-01-16:
Did you ensure that you’ve updated the AirSim plug-in for that environment?

sjnah, 2019-01-21:
I understood how to apply Airsim function to new environment as follows. Is that right?

1. Copy the airsim plugins file to the environment folder.
2. Run UE4Editor and compile it.
3. Change Gamemode to Airsim.

If that's true, I can do that with the AirSimNH environment available at the Unreal Marketplace. However, AirsimNH, which can be downloaded from Airsim-Release, does not have an uproject file. 
I know it can not be compiled because it can not be opened with UE4Editor.

rancheng, 2019-06-29:
> Weather effects such as rain and snow accumulation on surfaces has not been enabled in Neighborhood. With the latest AS code, you can see the rain/dust/fog/snow particle effects, but you won't see accumulations on surfaces without some changes to the materials.

Do you have any plan to add the surface snow or dust effect in the future? BTW, thanks for your hard work!

RVBldr, 2019-07-01:
That’s in work with a new Neighborhood environment. We just don’t have a firm release date set for that. Hopefully soon.

closed 2019-10-08 by msb336