microsoft/AirSim #1693: How to get started with airSim?

Supaple-x, 2019-01-11:
Hey guys!

I have 3d-model of environment and 3d-model of drone.

How can I use it in sim now? Is here step-by-step sequence how to behavior in the beginning to achieve the result?

Thanks a lot...

jasonhbartlett, 2019-01-11:
My suggestion would be to read the documentation contained in this repo perhaps starting with https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md if your platform is PC otherwise https://github.com/Microsoft/AirSim/blob/master/docs/build_linux.md if your platform is Linux.

Supaple-x, 2019-01-14:
Thanks, Jason,

but after I follow theese steps I'm getting error.
I've opened [another issue for that](https://github.com/Microsoft/AirSim/issues/1705)

closed 2019-01-14 by Supaple-x