microsoft/AirSim #1696: Unity has outdated data structure when parsing elements of the settings.json file

jasonhbartlett, 2019-01-11:
Unity has outdated data structure when parsing elements of the settings.json file.
To reproduce: 
When PX4 in settings file. 
```yaml
{
    "SettingsVersion": 1.2,
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SimMode": "Multirotor",
    "Vehicles": {
        "PX4": {
          "VehicleType": "PX4Multirotor"
        }
      }
}
```

[image]

I believe the structs located in AirSim/Unity/UnityDemo/Assets/AirSimAssets/Scripts/Utilities/AirSimSettings.cs script need to be updated to be able to read the "Vehicles" data structure.  These structs have been updated and no longer use the DefaultVehcileConfig element which appears to have been removed back on May 17th with the following merge:
53cdcde41a9c325ba34d06b02a45d960bdb543ca

I have noted a temporary work-around in Issue #1617

jasonhbartlett, 2019-02-07:
Just curious if there was any update on this?

jasonhbartlett, 2019-02-13:
FYI - I'm making progress on this and hopefully will have a pull request to submit soon.

jasonhbartlett, 2019-03-06:
Okay this was a **way** bigger task than I anticipated.  _However_... I've put some significant time into the effort and although there are still several items I need to clean up, I'm making good headway and hope to submit a pull request ... in a week?  The hard part about this is not having a full grasp of everything the AirLib/AirSimWrapper does as my C++ knowledge is pretty weak.  When I first started plugging away at it from the Unity side I was just thinking that the Unity client handled all the reading of the settings.json file.  After awhile I discovered the AirLibWrapper/AirLib also ingests it.  Sooo... okay well, I'll just keep at it on the Unity side and at the least the result will be that the Unity client will have a complete representation of the settings from the settings.json file at the outset.  I believe this will pave the way for immediately setting up the environment (time of day settings), the cameras, and the vehicles (pose) within the simulation.  And I believe based on the work I've done that this paves the way for multiple vehicles of the same type running in the environment together.
Anyhow, here is a link to the fork/branch and the vast bulk of the work can be found here:
[fix-unity-settings-file/UnityDemo/Assets/AirSimAssets/Scripts](https://github.com/jasonhbartlett/AirSim/tree/fix-unity-settings-file/Unity/UnityDemo/Assets/AirSimAssets/Scripts)
Particularly the **InitializeAirSim.cs**, **AirSimSettings.cs**, & **AirSimHUDScript.cs** in which the vehicle is now instantiated from prefab rather than having to be pre-placed in the scene.
Also, I got major help on the json stuff from a C# script created by Markus Göbel (Bunny83) called **SimpleJSON.cs** described as  a simple JSON parser that can build a JSON string from the node-tree, or generate a node tree from any valid JSON string and who added 2 functions (HasKey, and GetValueOrDefault) after an issue/question I created on his repo. [SimpleJSON](https://github.com/Bunny83/SimpleJSON) If this pull request gets approved I'll be sure to give him props.
Anyhow, if I can get the pull request in I'm sure there will be things to iron out before it is approved.  _I'm hoping to collab_ with someone who knows the AirLb/AirSimWrapper C++ side once this is kick-started.  Hopefully this will have been worthwhile effort and pave the way for multiple drones, etc.

ElCodeMonkey, 2019-03-21:
Any updates on this? It seems basic functionality of simply adding a Lidar to the vehicle causes it to crash. Maybe an older json format to follow for now? I tried parsing through the code to see where "Vehicles" or "Sensors" get used and I can't find anything within the Unity code. The existing structure doesn't seem to allow for any sensor modification so far as I can tell. Only option to add manually within Unity?

jasonhbartlett, 2019-03-25:
HI @ElCodeMonkey 
Unfortunately, as you say, it's all just still too new on the Unity side for why this stuff isn't there yet.  I too am surprised there hasn't more Unity community interest in this to at least bring the Unity capabilities up to the same as that is supported in Unreal.  I only started playing with AirSim at the same time that Unity support was released (experimentally) last October and at the time I was excited as I thought the Unity community would jump on this.  But efforts to improve the Unity (and actually everything AirSim) has been really slow going from what I can see.  Which is why I started working on a fix for the way Unity parses the settings.json file in order to bring it up to speed with the current settings file hierarchy.  I was getting close to submitting a pull request, but then some other things came up and I had to put it on hold.  Also, as I said, I don't have a full grasp of the AirLib C++ side of things, so I expect when I do submit a pull request, it will need some ironing out.  Hopefully I can get back to it soon.

ElCodeMonkey, 2019-03-26:
I thought I'd take a look at what you had so far and I guess I'm still missing some major concept here. I had mentioned that I didn't see where "Sensors" was even being looked at previously yet clearly it was being used by something since it made it crash. With your code, I do see it referenced and I see the settings object being set up with it, but I still don't see anyone actually using these settings to, say, create an actual Lidar GameObject. If I were to guess, I assume an AirSim client is launched which reaches into Unity to grab the settings and then looks for GameObjects by the names in the settings object as derived by the JSON. It probably expects certain objects to have certain parameters which the Unity code will need to have set up. If I'm right, this would beg the question what AirSim is looking for exactly. If I knew, I could perhaps take up where you left off. Is there documentation about that somewhere? Or perhaps a better starting point for how these things interface?

MelSoS, 2019-08-19:
any updates in this regard since March? I would be very interested as well!!!
Kind regards, MelSoS

ElCodeMonkey, 2019-08-19:
I essentially ditched it and went with LGSVL. In the end, I think it allows for more personalization for my particular needs. (And it works, too, of course which is a huge plus ;-))