microsoft/AirSim #1698: The gui says takeoff detected but It doesn't really move. The propellers are spinning though . How to fix it

CaoYongshengcys, 2019-01-12:
The gui says takeoff detected but It doesn't really move. The propellers are spinning though. Btw , the pixhawk is on altitude mode.

lovettchris, 2019-01-13:
When you say Pixhawk I assume this is HITL (hardware in the loop mode).  If so make sure you wait for GPS fusion to complete.  When you first connect you should see this output:
```
NFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: rootfs/fs/microsd/log
INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2019-01-13/01_55_46.ulg
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [mavlink] partner IP: 127.0.0.1
```
Then a pause for a few seconds while PX4 monitors GPS to ensure it is stable, then you should see this:
```
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
```
Now take off should work.  If you run "commander status" from a Pixhawk nsh shell then you should see this output:
```
pxh> commander status
WARN  [commander] type: symmetric motion
WARN  [commander] safety: USB enabled: [NO], power state valid:  [OK]
WARN  [commander] home: lat = 0.0000278, lon = 112.5548751, alt = 21.42, yaw: 0.02
WARN  [commander] home: x = 0.0016745, y = 0.0024613, z = 0.03
WARN  [commander] datalink: OK
WARN  [commander] main state: 11
WARN  [commander] nav state: 18
WARN  [commander] arming: STANDBY
```
which tells you it has a home location with z close to 0.  Checking this a few times with a few seconds apart will tell you if the altitude is stable.  You want be able to take off if alt or z is wondering around all over the place.   

Note: you can open a nuttx shell into the Pixhawk using this command:
```
d:\git\AirSim\MavLinkCom\MavLinkTest\build\x64\Debug\MavLinkTest.exe -server:localhost:14388
nsh
```
type "?" to see the list of px4 commands you can do and try "commander takeoff" for example and see if that works.  If that works then the pythonclient takeoff.py should also work.  Altitude control mode is:
```
PX4_CUSTOM_MAIN_MODE_ALTCTL
```
when you run commander take off the mode will automatically switch to:
```
    PX4_CUSTOM_MAIN_MODE_AUTO
    PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF
```
but if you try and change the mode to PX4_CUSTOM_MAIN_MODE_ALTCTL during take off then takeoff will be aborted.

CaoYongshengcys, 2019-01-13:
> When you say Pixhawk I assume this is HITL (hardware in the loop mode). If so make sure you wait for GPS fusion to complete. When you first connect you should see this output:
> 
> ```
> NFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
> INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
> INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
> INFO  [logger] logger started (mode=all)
> INFO  [logger] log root dir created: rootfs/fs/microsd/log
> INFO  [logger] Start file log
> INFO  [logger] Opened log file: rootfs/fs/microsd/log/2019-01-13/01_55_46.ulg
> INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
> INFO  [mavlink] partner IP: 127.0.0.1
> ```
> Then a pause for a few seconds while PX4 monitors GPS to ensure it is stable, then you should see this:
> 
> ```
> INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
> INFO  [ecl/EKF] EKF commencing GPS fusion
> ```
> Now take off should work. If you run "commander status" from a Pixhawk nsh shell then you should see this output:
> 
> ```
> pxh> commander status
> WARN  [commander] type: symmetric motion
> WARN  [commander] safety: USB enabled: [NO], power state valid:  [OK]
> WARN  [commander] home: lat = 0.0000278, lon = 112.5548751, alt = 21.42, yaw: 0.02
> WARN  [commander] home: x = 0.0016745, y = 0.0024613, z = 0.03
> WARN  [commander] datalink: OK
> WARN  [commander] main state: 11
> WARN  [commander] nav state: 18
> WARN  [commander] arming: STANDBY
> ```
> which tells you it has a home location with z close to 0. Checking this a few times with a few seconds apart will tell you if the altitude is stable. You wont be able to take off if alt or z is wondering around all over the place.
> 
> **Note:** you can open a nuttx shell into the Pixhawk using this command:
> 
> ```
> d:\git\AirSim\MavLinkCom\MavLinkTest\build\x64\Debug\MavLinkTest.exe -server:localhost:14388
> nsh
> ```
> type "?" to see the list of px4 commands you can do and try "commander takeoff" for example and see if that works. If that works then the pythonclient takeoff.py should also work. Altitude control mode is:
> 
> ```
> PX4_CUSTOM_MAIN_MODE_ALTCTL
> ```
> but when you run `commander take` off the mode will automatically switch to:
> 
> ```
>     PX4_CUSTOM_MAIN_MODE_AUTO
>     PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF
> ```
> but if you try and change the mode to PX4_CUSTOM_MAIN_MODE_ALTCTL during take off then takeoff will be aborted.

> When you say Pixhawk I assume this is HITL (hardware in the loop mode). If so make sure you wait for GPS fusion to complete. When you first connect you should see this output:
> 
> ```
> NFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
> INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
> INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
> INFO  [logger] logger started (mode=all)
> INFO  [logger] log root dir created: rootfs/fs/microsd/log
> INFO  [logger] Start file log
> INFO  [logger] Opened log file: rootfs/fs/microsd/log/2019-01-13/01_55_46.ulg
> INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
> INFO  [mavlink] partner IP: 127.0.0.1
> ```
> Then a pause for a few seconds while PX4 monitors GPS to ensure it is stable, then you should see this:
> 
> ```
> INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
> INFO  [ecl/EKF] EKF commencing GPS fusion
> ```
> Now take off should work. If you run "commander status" from a Pixhawk nsh shell then you should see this output:
> 
> ```
> pxh> commander status
> WARN  [commander] type: symmetric motion
> WARN  [commander] safety: USB enabled: [NO], power state valid:  [OK]
> WARN  [commander] home: lat = 0.0000278, lon = 112.5548751, alt = 21.42, yaw: 0.02
> WARN  [commander] home: x = 0.0016745, y = 0.0024613, z = 0.03
> WARN  [commander] datalink: OK
> WARN  [commander] main state: 11
> WARN  [commander] nav state: 18
> WARN  [commander] arming: STANDBY
> ```
> which tells you it has a home location with z close to 0. Checking this a few times with a few seconds apart will tell you if the altitude is stable. You wont be able to take off if alt or z is wondering around all over the place.
> 
> **Note:** you can open a nuttx shell into the Pixhawk using this command:
> 
> ```
> d:\git\AirSim\MavLinkCom\MavLinkTest\build\x64\Debug\MavLinkTest.exe -server:localhost:14388
> nsh
> ```
> type "?" to see the list of px4 commands you can do and try "commander takeoff" for example and see if that works. If that works then the pythonclient takeoff.py should also work. Altitude control mode is:
> 
> ```
> PX4_CUSTOM_MAIN_MODE_ALTCTL
> ```
> but when you run `commander take` off the mode will automatically switch to:
> 
> ```
>     PX4_CUSTOM_MAIN_MODE_AUTO
>     PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF
> ```
> but if you try and change the mode to PX4_CUSTOM_MAIN_MODE_ALTCTL during take off then takeoff will be aborted.

Thank you, I have fixed this problem.

closed 2019-01-22 by madratman

CaoYongshengcys, 2019-03-18:
Reinstall the software and the version should be within the scope of approval

machenxiang, 2019-03-18:
@CaoYongshengcys hi caoyongsheng .can you
tell me the the version of airsim and px4 firmware. thank you any way.兄弟能告诉我你的airsim和px4固件 ue4具体版本吗？