microsoft/AirSim #1703: Recording from multiple cameras not supported in unity

adrshm91, 2019-01-13:
I am trying to save images using multiple cameras using the record button in the GUI. I have added the camera names in the settings.json but the function GetCameraCaptureForRecording in VehicleCompanion.cs takes only one camera into consideration. Is this feature yet to be added

msb336, 2019-01-17:
@adrshm91 please attach you `settings.json` file to this thread so I can reproduce your issue

tjxjb, 2019-04-02:
I'm also having this problem, only seems to save the first CameraName's image. Using Unity AirSim, with the Windridge map. Using same settings as used in Unreal (and it works, records all 3 images like I want). Settings are as follows:

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "ViewMode": "",
  "ClockSpeed": 1.5,
  "Recording": {
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "0", "ImageType": 3, "PixelsAsFloat": false, "Compress": false },
        { "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": false },
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": false }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 600,
        "Height": 300,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 10,
        "AutoExposureBias": 0
      },
      {
        "ImageType": 5,
        "Width": 600,
        "Height": 300

      },
      {
        "ImageType": 3,
        "Width": 600,
        "Height": 300

      }
    ]
  },
  "SubWindows": [
    {"WindowID": 0, "CameraName": "0", "ImageType": 3, "Visible": false},
    {"WindowID": 1, "CameraName": "0", "ImageType": 5, "Visible": false},
    {"WindowID": 2, "CameraName": "0", "ImageType": 0, "Visible": true}    
  ],
  "DefaultSensors": {
        "Barometer": {
             "SensorType": 1,
             "Enabled" : true
        },
        "Gps": {
             "SensorType": 3,
             "Enabled" : true
        },
	"Imu": {
             "SensorType": 2,
             "Enabled" : true
        },
        "Lidar1": { 
             "SensorType": 6,
             "Enabled" : false,
             "NumberOfChannels": 16,
             "PointsPerSecond": 10000
        }
    }
}