microsoft/AirSim #1707: HITL - Drone keeps velocity even after changing directions

hartmann100, 2019-01-15:
I have a HITL setup with a PX4 Flight Controller, AirSim 1.10 and a custom unreal project (4.19.3) and (for the most part) it works fine. My problem is, that the drone flight does not feel realistic. I have two problems that maybe correlate:
- The drone contains velocity and i have to steer against that to come to an halt. Or in other words: When i tilt the drone to the right side it moves right and when I tilt the drone back to a straight position it retains its velocity and keeps moving right.
- In manual mode where the drone should stabilize to a straight position it sometimes stabilizes to a slightly tilted position. This change occurs when I tilt the drone very hard to one side. If this new "saved stabilized" position is tilted to the left, I have to tilt the drone very hard to the right for a second to put the saved position to somewhat where it was before.

I am not quite sure where to look for the problem/solution and appreciate any help.

lovettchris, 2019-01-15:
Number 1 is just normal momentum - to cancel right movement you have to go back more than level you have to tilt left to cancel the rightwards movement.  Now in a real drone the propellers do actually act as a bit of a brake on lateral movement, and AirSim is not fully simulating all those complex effects which makes it feel like the drone has more momentum than normal.  You can also play with the drone's simulated "mass" which has a big effect on this.

I have seen your number 2 bullet also, and do not know a good fix.  I suspect PX4 is trying to compensate for the IMU drift that happens with real IMU's when  you tilt the drone for extended periods of time, but I have not found a way to turn that off.  I will ping the PX4 forums and see if we can find a solution to this long standing problem.  In the meantime take a look at the PX4 EKF2 position estimator, if you see any controls over drift compensation try turning them off and see what happens.

hartmann100, 2019-01-23:
Ah, thank you for your answer! (And sorry for my long delay.)
I tried to play a little bit with the simulated mass and got it a little bit better. But I still have two problems when changing my drone specs: First the drone falls very slowly (even when the rotors are at 0% thrust) and i am not quite sure where i can change that. When I increase the mass i have to increase the thrust too, but there is my second problem: If i increase the power to much (> 30%) the drone starts jerking very fast. It is the same if i increase the clockspeed to more than 2. Soo... i am will keep playing with the settings.

To the second bullet: For me it is good to know, that it is a seperate problem. I will look at the PX4 EKF2
position estimator, if i find the time.

timesofbadri, 2019-03-09:
AirSim behaves the way it is supposed to be. As you may already know, the default UAV from AirSim uses Imu, Magnetometer, Gps & Barometer. 

Simulation can be done in many ways. These are setups that you can try.
1. AirSim and Xbox controller. Navigation is done by using AirSim based API's.
2. SITL using PX4. Navigation can be done by using either the API's from AirSim or from PX4 (MAVLink based).
3. HITL using the original Pixhawk (not the CUBE version)  Navigation is taken care of by the Pixhawk. You can play around with EKF2/Quaternion/EKF3 as you may like. AirSim/Unreal is just the rendering engine.
4. You may use the above setup (1 through 4) and play around with indoor and outdoor based environments. Indoor environments don't and shouldn't depend on GPS for navigation (in an ideal scenario).

So, for the start, eliminate if its hardware setting or software setting. I suggest, try setup # 1 first and then eliminate with what you have. If your setup works really well for setup # 1 as per their documentation, then it should work fine for the rest. 

Firmware matters a lot when it comes to HITL. The firmware 1.5.5 ([https://github.com/PX4/Firmware/releases/tag/v1.5.5]) works for me in HITL. @lovettchris is spot on. The drift that the UAV has to handle is really hard to simulate when it comes to Manual mode. This is the reason the hardware companies are investing in making dual IMU's and other supporting sensors for navigation more accessible. 

"First the drone falls very slowly (even when the rotors are at 0% thrust) and i am not quite sure where i can change that. When I increase the mass i have to increase the thrust too, but there is my second problem: If i increase the power to much (> 30%) the drone starts jerking very fast. It is the same if i increase the clockspeed to more than 2" Ahh man, it looks to me like you are trying to solve a hardware problem using HITL and AirSim. ;) :) Why don't you try the normal Gazebo instead? Sorry to offtrack you. But it will solve the problem.

lovettchris, 2019-04-07:
Not sure I agree with the points timesofbadri is making.  The PX4 software supports a special simulation mode called HITL (hardware in the loop) and when in this mode it actually tuns off all those real hardware sensors (imu, mag, gyro, gps and baro) and instead it gets these sensor readings from AirSim over the mavlink connection.  The PX4 then does it's normal "flying" algorithms based on that and returns motor power outputs back to AirSim.  AirSim applies the appropriate thrust to each motor and simulates the result of that by sending back updated sensor readings, etc.  Why would someone go to all this trouble?  Because provides a way to test the real PX4 "flight controls" with your specific python code in a Simulator first before trying it on a real drone.  Sure you could instead test it all in SITL mode, but what if SITL and HITL behave differently?  Then you would not catch that bug until it is too late and your expensive drone crashes.  But as always there's no free lunch.  Getting HITL mode to work well is non-trivial.

hartmann100, 2019-08-30:
Thank you for helping.
I will close this issue for now, because I dont have that much time working on it right now.

closed 2019-08-30 by hartmann100