microsoft/AirSim #1710: There is something wrong with \AirSim\PythonClient\car\DQNcar.py

CaoYongshengcys, 2019-01-15:
My error is as follows, anyone can help me ?

(base) D:\Unreal Projects\AirSim\PythonClient\car>python DQNcar.py
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Selected CPU as the process wide default device.
Traceback (most recent call last):
  File "DQNcar.py", line 536, in <module>
    current_state = transform_input(responses)
  File "DQNcar.py", line 434, in transform_input
    img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
  File "D:\Anaconda3\lib\site-packages\numpy\core\fromnumeric.py", line 257, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "D:\Anaconda3\lib\site-packages\numpy\core\fromnumeric.py", line 52, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 1 into shape (0,0)

srivatsankrishnan, 2019-01-17:
I am also running into the same issue but with my own example. Also it happens pretty randomly.
My stack trace is same as soon as it hits numpy call:
```
img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
  File "/home/wxxx/.local/lib/python2.7/site-packages/numpy/core/fromnumeric.py", line 279, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "/home/wxxx/.local/lib/python2.7/site-packages/numpy/core/fromnumeric.py", line 51, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 1 into shape (0,0)
```

srivatsankrishnan, 2019-01-18:
I think its a bug in Airsim client.simGetImages API. The issue is the responses[0].height and responses[0].width returns 0. So the numpy reshape errors out once it encounters that.

The workaround I have is to check for the condition where the responses[0].width and responses[0].height is "0" and once that condition occurs, I pass dummy image ( all 1's) and then reset airsim so it doesn't crash simulation.

Ahmad5112, 2019-02-01:
@CaoYongshengcys did you resolve the issue? If so how did you resolved

CaoYongshengcys, 2019-02-15:
> @CaoYongshengcys did you resolve the issue? If so how did you resolved

I did not solve this problem

annisanazi, 2020-02-13:
did anyone resolve this issue?

TomAvrech, 2020-10-01:
Getting the same problem :(

zimmy87, 2021-01-07:
possibly related to #1009

zimmy87, 2021-01-15:
Hi, thank you for reporting this issue.  This script was recently removed in #3215 and replaced with a new script at \AirSim\PythonClient\reinforcement_learning\dqn_car.py that uses the new airgym module. I tested the new script locally and it runs for me. Can you retest with the new script and report back if you see any issues?

closed 2021-01-15 by zimmy87