microsoft/AirSim #1711: Is it possible to manually specify the communication port?

flyers, 2019-01-15:
Is it possible to manually specify the communication port between the server and the python client?
I want to do this so that I can start multiple instances of the environment and have multiple python clients, each communicating with the corresponding environment instances without affecting each other.

Thanks.

DavidLSmyth, 2019-01-15:
I don't think there's a trivial way to do it (but I'm not sure since I've been working off an older version for a few months) since airsim will try and read the settings file from Documents/AirSim/settings.json. Off the top of my head I can think of a few lazy ways that might work to get around it though:

1. Pull an older version that contains  "ApiServerPort" in settings.json (e.g. the 4.16 branch) and then run an instance of airsim. It will read settings.json and begin running with the specified settings. Once the first instance is running, in the settings.json file change "ApiServerPort": 41451 to "ApiServerPort": 41452 and run a second instance of AirSim. You will need to also change the port the python client connects to (very straightforward). 

2). In the master branch, by default the RPC client will begin running on 41451. If you have space on your drive git clone into multiple folders the latest master branch. Then replace https://github.com/Microsoft/AirSim/blob/5223c26f86d93b7a3b5961dab170a829a9d5baab/AirLib/include/api/RpcLibServerBase.hpp#L17 with ` RpcLibServerBase(ApiProvider* api_provider, const std::string& server_address, uint16_t port = 41451);` `  RpcLibServerBase(ApiProvider* api_provider, const std::string& server_address, uint16_t port = 41452);`, etc. in each airsim folder respectively. Do same for https://github.com/Microsoft/AirSim/blob/5223c26f86d93b7a3b5961dab170a829a9d5baab/AirLib/include/api/RpcLibClientBase.hpp#L23 and other RPCLib files (although I think apis still inherit from these so might not be necessary)

3). Run multiple instances on separate virtual machines using virtualbox or something similar (this would be terrible for hard disk memory)

4). Try and use azure VMs as mentioned here: https://github.com/Microsoft/DroneRescue. 

5). Modify the source code so that you can pass in the api port and settings file.

I'm not sure if any of these will work but maybe they'll help get you started. Hopefully someone will come up with a far cleaner solution for you

flyers, 2019-01-16:
@DavidLSmyth Thanks for your helpful suggestions! I will try some of them.

bybunni, 2019-03-20:
I was able to accomplish this pretty trivially by modifying AirSimSettings.hpp to read in a port I specified in the settings.json and then modifying SimModeCar.cpp and SimModeWorldMultirotor.cpp to pass that specified port when creating the API server.