microsoft/AirSim #1712: Car segmentation problem in City Environment

ximenesfel, 2019-01-15:
Hi,

My setup:
OS: Windows 10
Airsim: v.1.2.0
Unreal version: 4.18

I'm using Airsim for gathering data. I will use this data to train a deep learning network. In the neighbor environment, I was able to obtain car segmentation. Bellow a mask obtained.
[image]
[image]

But when I tried to do the same in City environment some issues happens. In some places the car segmentation not appear and for moving cars  the segmentation is not showed (I don't know if this is a problem or this feature is not developed. For pedestrian moving is possible to obtain segmentation.) 

I recorded a video showing this issue. Sorry for my terrible drive, but I started using RC today, and I experienced some difficulties. But I think you can see the problems that I mentioned.

https://drive.google.com/file/d/1bShqStFYX2QUCLh-2Et5MRYTa3PZ6AOc/view?usp=sharing

msb336, 2019-01-16:
@ximenesfel This issue is because the cars in AirSimNH are static meshes, whereas the cars in CityEnv are actors in the scene. You'll notice you don't see pedestrians in the city environment either. Currently airsim doesn't look for actors in the segmentation process, but hang tight, we're working to fix that now!

ximenesfel, 2019-01-17:
@msb336 Thanks for your explanation. Great to know that you're working to fix that now. :)

ElliotHYLee, 2019-03-04:
Is this fixed?

gjy3035, 2019-07-09:
The newest version can obtain the moving cars and pedestrians?

AloshkaD, 2019-09-22:
Not fixed for me in the latest version.

mostafa172, 2020-03-31:
any updates on this issue, is it fixed yet?