microsoft/AirSim #1714: Drone can't hover in Windridge City (Unity)

FurqanHabibi, 2019-01-17:
I managed to add the drone to windridge city using steps from [here](https://github.com/Microsoft/AirSim/blob/master/docs/custom_unity_environments.md) and [here](https://github.com/Microsoft/AirSim/issues/1644)

When I tried flying the drone using the python client, the drone will fly and arrive at the destination but then fail to keep it's position (hover) and just fall slowly as can be seen in the following video :

[image]

Does anyone knows why this is and any way to fix it?

Thank you.

jasonhbartlett, 2019-01-17:
Hi @FurqanHabibi 
I am not seeing that behavior. Although I have my own issues with WindRidgeCity, namely that on startup the drone falls right through the ground rather than be "caught" by the collision with the box collider.  The only thing that seems to work is if I start with the Drone at a position really really high and give it time to fall before hitting the ground AND making sure the collider of that object it hits is really really big.  I have also tried adding a script that holds off on enabling the Quadrotor and AirSimHUD game objects until after the scene has fully loaded, (or even 5 seconds after the scene loaded).  I did this because I thought maybe it was loading and started the physics calculation for those objects before the environment had time to finish loading.  But alas that didn't do anything.
I also tried setting gravity to 0.0 vs 9.81  -   Again, no effect.

Anyhow, at least in my scenario where I CAN get the drone to stop when it hits the ground and be ready for me to start the python script.  I AM seeing the drone maintain it's position after the last moveToPositionAsync() call.  Here is the script I used to test it out:
```python
import sys
import time

import setup_path 
import airsim

def print_position(client):    
    state = client.getMultirotorState()
    gps = state.gps_location
    print("gps location lat={}, lon={}, alt={}".format(gps.latitude, gps.longitude, gps.altitude))
    pos = state.kinematics_estimated.position
    print("local position x={}, y={}, z={}".format(pos.x_val, pos.y_val, pos.z_val))

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

print_position(client)

landed = client.getMultirotorState().landed_state
if landed == airsim.LandedState.Landed:
    print("taking off...")
    client.takeoffAsync().join()
else:
    client.hoverAsync().join()

pos = client.simGetGroundTruthKinematics().position
z = pos.z_val - 7
client.moveToPositionAsync(pos.x_val,pos.y_val,z,1).join()
```

jasonhbartlett, 2019-01-17:
Hi @FurqanHabibi 
Here is a short [video](https://youtu.be/B34Em3lo35A) to demonstrate.

FurqanHabibi, 2019-01-18:
Hello @jasonhbartlett 
Thank you for your help! I think I found the culprit.
So it looks like the drone will behave like my video above if I set the position to be round numbers, like so
```
client.moveToPositionAsync(-210, 262, -352, 5).join()
```
or
```
pos = client.simGetGroundTruthKinematics().position
z = pos.z_val - 10
client.moveToPositionAsync(int(pos.x_val), int(pos.y_val), int(z), 5).join()
```

However if I set the position to be just like in your code
```
pos = client.simGetGroundTruthKinematics().position
z = pos.z_val - 7
client.moveToPositionAsync(pos.x_val, pos.y_val, z, 1).join()
```
it will fly and hover successfully.

I still don't know the reason and maybe someone from the AirSim team can explain, but at least now I can successfully fly the drone in windridge.

About the drone falling through the ground right after starting the simulation, I also encounter the exact same thing. And I get over it in the exact same way as you, namely starting the drone in a high enough position. Maybe someone have a better way to do it?