microsoft/AirSim #1715: Does the camera intrinsic parameters change as the distance between the camera and the subject changes?

Capr1corn, 2019-01-17:
OK, as expected the MATLAB Stereo Camera Calibration App had problems auto-detecting the checkerboard if there are similar objects in the background (e.g. the blocks from the AirSim Environment 'Blocks'). I removed them et voila, all pictures were taken into account for calibration:

[image]

I got the following results:

## Left Camera (#1)
### Intrinsic Matrix
571,402636917134	0	0
-0,0833452467134678	571,544739332093	0
400,639999395168	300,427684345250	1
### Focal Lenght
571,402636917134	571,544739332093
### Principal Point
400,639999395168	300,427684345250
### Skew
-0.083345246713468
### Radial Distortion
0,000415036899996763	0,000556254050889156	-0,00275943702139659
### Tangential Distortion
-0,000101728992258693	0,000205945187867511

## Right Camera (#2)
### Intrinsic Matrix
571,222935812704	0	0
-0,0184437475149830	571,365120227730	0
400,363512365099	300,369927037223	1
### Focal Lenght
571,222935812704	571,365120227730
### Principal Point
400,363512365099	300,369927037223
### Skew
-0.018443747514983
### Radial Distortion
0,00204149753440190	-0,00666695925778359	0,00559941924299634
### Tangential Distortion
-9,25453504591489e-05	-7,74448998384522e-05

## Stereo Camera Relation (#2 relative to #1)
### Rotation
0,999999877459772	1,16767450105424e-07	-0,000495055983578262
-1,62889434791495e-07	0,999999995660114	-9,31651505587194e-05
0,000495055970551118	9,31652197816300e-05	0,999999873119906
### Translation
-124,610118434116	0,00529493336163922	-3,87205255315630

_Originally posted by @JonathanSchmalhofer in https://github.com/Microsoft/AirSim/issues/269#issuecomment-350748910_

When I was doing camera calibration, I deliberately calibrated the calibration plate at different distances, and then found that the camera's internal parameters fx, fy will change with the distance between the camera and the subject.
Can someone explain why this happened?

Capr1corn, 2019-01-17:
At the time of calibration, I fixed the calibration plate at coordinates (1350, 888, 250). For the first calibration, the coordinates of the camera are (0, 888, 225). The first camera calibration gives the camera internal parameters:
[902.3558976965726, 0, 321.3648554626281;
 0, 900.0096033547145, 204.8016562704659;
 0, 0, 1]
For the second calibration, the coordinates of the camera are (-100, 888, 225), and the second camera calibration gives the camera internal parameters:
[876.2327324144362, 0, 321.9009847334212;
 0, 874.0885350641621, 206.2475806037728;
 0, 0, 1]
For the third calibration, the coordinates of the camera are (-300, 888, 225), and the third camera calibration gives the camera internal parameters:
[773.0785881016429, 0, 322.1147723437926;
 0, 769.9143252388714, 208.6918739775566;
 0, 0, 1]

closed 2019-01-21 by Capr1corn

Carl-Yan, 2019-12-13:
Hi~I was also confused by this issue, could you tell me the reason of this parameters floating?And, how to stabilize the parameters? Thanks~ : )

ningshipang, 2020-07-25:
I also want to make camera calibration, but I don't know how to put the picture used to make camera calibration in the background. Could you tell me the way? Thank you!