microsoft/AirSim #1724: Segmentation Image API bug when in headless mode of NoDisplay setting

flyers, 2019-01-21:
I found that in the latest master branch, the `simGetImages` API returns wrong image types if we use `"NoDisplay"` viewmode in `settings.json`. I use the Blocks environment and the following python scripts to reproduce this bug. When the viewmode is `"NoDisplay"`, the following code gets the RGB image instead of the segmentation image.

```python
import numpy as np
import cv2
import airsim

def uncompressed_to_cvimage(response):
    '''
    Convert uncompressed RGBA image to cv2 BGR format H x W x Channels
    '''
    assert response.pixels_as_float == False
    img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8)
    img = img1d.reshape(response.height, response.width, 4)[:, :, 0:-1]
    return img[:, :, ::-1]

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

response = client.simGetImages([airsim.ImageRequest(
    "0", airsim.ImageType.Segmentation, False, False)])[0]
seg = uncompressed_to_cvimage(response)
cv2.imshow('seg', seg)
cv2.waitKey(0)
```

mfe7, 2019-02-20:
I have also observed this behavior.

AndreiBaraian, 2019-03-14:
I have observed the same issue, including depth and depth perspective type of images, they are just RGB images when AirSim runs in "NoDisplay" mode. Also, not only the simGetImages() API works like that, but also when I try to record by pressing the "R" key.

dimikout3, 2020-01-22:
I observed the same behavior for depth images. I am using a swarm of 4 drones and when the "NoDisplay" is activated I can not retrieve the depth image for one drone, however I can retrieve the depth images for the other 3 drones, which is really weird behavior.

saihv, 2020-01-22:
@dimikout3 Do you receive any other image (such as RGB) for the fourth drone or none at all?

dimikout3, 2020-01-23:
I am looking at it right now, I am actually retrieving both RGB and Depth images. RGB is perfect, on the contrary depth image is kind of scaled down, between vallues of [0.0, 0.97]. I am not causing that scalling, for the other drones these values are [12.5, 1025.0], which makes sense. Attached are the images (Drone1 is the "problematic" one)
[image]
[image]
[image]
[image]

ironclownfish, 2020-02-06:
note to us: the fix for the IR images in the Africa scene might be relevant.

sourabhlal, 2020-07-15:
Does anyone have a solution for this issue? I'm not getting the right segmentation image either in NoDisplay mode

Zartris, 2021-03-23:
Did not solve this issue today, but we found a workaround.
The problem is when you use the default camera "0" to get the segmentation from.
An alternative (workaround) is to create your own front camera:
```
{
	"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",
	"ClockSpeed": 1,
	"ViewMode": "NoDisplay",
	"Vehicles": {
		"SimpleFlight": {
			"VehicleType": "SimpleFlight",
			"DefaultVehicleState": "Armed",
			"AutoCreate": true,
			"PawnPath": "",
			"EnableCollisionPassthrogh": false,
			"EnableCollisions": true,
			"AllowAPIAlways": true,
			"RC": {
				"RemoteControlID": 0,
				"AllowAPIWhenDisconnected": false
			},
			"Cameras": {
				"cam": {
					"CaptureSettings": [
						{
							"ImageType": 0,
							"Width": 240,
							"Height": 180,
							"FOV_Degrees": 90,
							"AutoExposureSpeed": 100,
							"AutoExposureBias": 0,
							"AutoExposureMaxBrightness": 1,
							"AutoExposureMinBrightness": 1,
							"MotionBlurAmount": 0,
							"TargetGamma": 1.5
						},
						{
							"ImageType": 5,
							"Width": 240,
							"Height": 180,
							"FOV_Degrees": 90,
							"AutoExposureSpeed": 100,
							"AutoExposureBias": 0,
							"AutoExposureMaxBrightness": 1,
							"AutoExposureMinBrightness": 1,
							"MotionBlurAmount": 0,
							"TargetGamma": 1.5
						}
					],
					"X": 0.4,
					"Y": 0.0,
					"Z": 0.0,
					"Pitch": 0.0,
					"Roll": 0.0,
					"Yaw": 0.0
				}
			}
		}
	}
}
```
Then request the images from this camera instead of the "0".
This should work in both "NoDisplay" and other modes.

PR #4147 (2021-12-01): Sets bCapture* true only while showing subwindow 
Sets bAlwaysPersistRenderingState to true. This fixes bugs related to capture while on NoDisplay.
Sets bCaptureOnMovement and bCaptureEveryFrame to true only when they are going to be viewed on a subwindow

Fixes: #1724 
Fixes: #2967
Fixes: #3504
Fixes: #3070

Partially fixes #4112

Should fix most of the FPS while capturing problems.

## About

This PR fixes the NoDisplay/Headless bug by setting bAlwaysPersistRenderingState to true.

Setting bCaptureOnMovement and bCaptureEveryFrame to true only while it's on an active subwindow camera improve [... cut]
closed 2021-12-01 by zimmy87