microsoft/AirSim #1726: Any API for reading input from joystick?

alexanderkoumis, 2019-01-22:
I would like to create a dataset consisting of images, timestamps, ground truth position/orientations, as well as thrust, roll, pitch, yaw input from my RC. The exact values I would like are visible on the main screen as "Joystick (T,R,P,Y Buttons): %f, %f, %f, %f". I tried reading this information in Python by calling:
```python
client = airsim.MultirotorClient()
rc_data = client.confirmConnection().rc_data
```
However, all this information is uninitialized:
```
<RCData> {   'is_initialized': False,
    'is_valid': False,
    'left_z': 0.0,
    'pitch': 0.0,
    'right_z': 0.0,
    'roll': 0.0,
    'switches': 0,
    'throttle': 0.0,
    'timestamp': 0,
    'vendor_id': '',
    'yaw': 0.0}
```
Is this struct only populated when the device is being controlled using the API? Is there another Python command I need to call to retrieve the information when controlling the quad with a RC?

Thanks for your time.

alexanderkoumis, 2019-01-22:
This seems to be where rc_data needs to be populated (in MultirotorApiBase.hpp):
```c++
    MultirotorState getMultirotorState() const
    {
        MultirotorState state;
        state.kinematics_estimated = getKinematicsEstimated();
        //TODO: add GPS health, accuracy in API
        state.gps_location = getGpsLocation();
        state.timestamp = clock()->nowNanos();
        state.landed_state = getLandedState();
        state.rc_data = getRCData();
        return state;
    }
```
However it seems like getRCData() is not implemented for the child class (I think MavLinkMultirotorApi.hpp?) so it defaults to the invalid RCData struct in VehicleApiBase.hpp. This is a very large project and I have only just begun looking at the code so I do not have a good mental model for how all the interfaces fit together. My goal is for the line above: `state.rc_data = getRCData();` to call `PawnSimApi::getRCData()` (which is populated with the correct values) instead of `VehicleApiBase::getRCData()`. I would appreciate pointers on how to accomplish this plumbing.

Thank you.

alexanderkoumis, 2019-01-26:
@hesham-akmal thanks for the tip, this is the direction I went in. Read the data in with pygame, and replayed them with https://github.com/tidzo/pyvjoy. Works okay, but not perfect, there seems to be some type of drift.

Would be nice to access the data AirSim is using as there is some normalization between vjoy -> QGroundControl -> AirSim, so it would be nice to grab the processed data with `client.getMultirotorState().rc_data`, train a model, and have it estimate it for input using `setRCData()`, instead of estimating output for pyvjoy interface.

HymanMurphy, 2019-11-19:
did you have resolved this issue?

ironclownfish, 2020-02-06:
Looks like we probably need to update this struct under more general circumstances -- this is a bug.

gloriazhang8664, 2021-11-09:
I was able to modified the PawnSimApi.cpp file to output the rc data to the airsim_rec file where all other data is stored whenever a recording is saved.