microsoft/AirSim #1729: Map customization with Roadrunner from Vectorzero

Atlunch, 2019-01-22:
Hello everyone,

is it possible to create your own map?
Does the map customization based on opendrive formats?
Would it be possible to use the road generation tool "Roadrunner" from vectorzero to create your own map?
https://www.vectorzero.io/

best regards
Atlunch

jasonhbartlett, 2019-01-23:
I don't know much about opendrive, but this VectorZero company website states that they can export to Unreal, Unity, or .fbx (which works in Unreal or Unity), so I'm guessing this could be used like a model generation tool and the models can be imported to the game engine and then you can do whatever else you want from there.  I don't know how much their product costs, but keep in mind that for decent prices and sometimes free, you can find roads, buildings, and all kinds of environment props in the Unity asset store or Unreal marketplace.   For example: [EasyRoads3d](https://assetstore.unity.com/packages/tools/terrain/easyroads3d-pro-v3-469)

bhappy10, 2019-01-28:
I made my own map using 'RoadRunner'.
If you make your map using 'RoadRunner', you see the video:
https://www.youtube.com/watch?v=OcbYuTFAo8o&t=2s

This picture is my map apply for AirSim
[image]

[image]

msb336, 2019-01-29:
It looks like you have solved this issue on your own. If you are still having problems, please re-open the thread.

closed 2019-01-29 by msb336