microsoft/AirSim #1731: Jupyter stuck in * when calling car client

wonjoonSeol, 2019-01-23:
Hello,

Everytime I call car client, the cell completes its execution as can be seen from the print statement, but the following cells are stuck in astericks.

[image]

This is one example, but it happens to all other cells that uses car client API.

Windows 10, Airsim v1.2.1 - Windows binary (Not built), python 3.6.8,  jupyter 4.4.0

madratman, 2019-01-23:
@wonjoonSeol in general, instead of a screenshot, please post python code, for ease in reproduction of your issue. 
I think the script is stuck as `plt.imshow()` is a blocking call. you can simply comment the imshow() line, or better still either pass the non-blocking parameter, or use the interactive mode, as explained [here](https://stackoverflow.com/questions/28269157/plotting-in-a-non-blocking-way-with-matplotlib)

closed 2019-01-23 by madratman

wonjoonSeol, 2019-01-30:
Thank you for the comment, 
plt was not the cause of the issue here.

Share.scripts_downpour.app.airsim_client from DistributedRL guide that overriden CarClient call was the issue.

[AutonomousDrivingCookbook/issues/85](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/85)

danialvi, 2022-03-18:
> 

Hi. How did you solve this. I tried deleting the DistributedRL folder, so it is not overriden the E2Edeeplearning. But still stuck after connection is establish. Please help! Been trying to solve this issue for several weeks