microsoft/AirSim #1732: Recording of self-defined camera images

jonasgrimm, 2019-01-23:
We are trying to record images through self-defined cameras. We edited the settings.json file as follows:

{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "Vehicles": {
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "RC": {
        "RemoteControlID": -1
      },
      "Cameras": {   
        "MyCamera0": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 0, "Roll": 0, "Yaw": -25
        },
        "MyCamera1": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 0, "Roll": 0, "Yaw": 0
        },
		"MyCamera2": {
          "X": 2, "Y": 0, "Z": -1,
		  "Pitch": 0, "Roll": 0, "Yaw": 25
        }
      }
    }
  },
  "SubWindows": [
	{"WindowID": 0, "ImageType": 0, "CameraName": "MyCamera0", "Visible": true},
    {"WindowID": 1, "ImageType": 0, "CameraName": "MyCamera1", "Visible": true},
    {"WindowID": 2, "ImageType": 0, "CameraName": "MyCamera2", "Visible": true}
  ],
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
		{ "CameraName": "MyCamera0", "ImageType": 5, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "MyCamera1", "ImageType": 5, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "MyCamera2", "ImageType": 5, "PixelsAsFloat": false, "Compress": true }
    ]
  }
}

When running AirSim, the images in the subwindows are depicted as wanted. However, when trying to record those images, a fatal error occurs.

Muntabia, 2019-01-24:
Hey @madratman, thanks for the quick response! 
After your comment we tried the same settings at another PC and it also worked. So it seems to be an unique problem.
We reset the Airsim installation at the first PC and now it is working.
For the sake of completeness I append the log file with the error and we are running the neighborhood env.

Once again, thank you!

`Log file open, 01/24/19 10:25:39
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VSPerf140.dll' (GetLastError=126)
LogWindows: File 'VSPerf140.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(-1873173895) SRandInit(-1873173894).
LogTaskGraph: Started task graph with 5 named threads and 8 total threads with 3 sets of task threads.
LogStats: Stats thread started at 1.878556
LogD3D11RHI: Aftermath initialized
LogPluginManager: Mounting plugin AirSim
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin HTML5Networking
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AppleARKit
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogInit: Using libcurl 7.47.1
LogInit:  - built for x86_64-pc-win32
LogInit:  - supports SSL with OpenSSL/1.0.2g
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_IDN
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = false  - Libcurl will NOT verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = nullptr  - Libcurl will use whatever was configured at build time.
LogInit: Build: ++UE4+Release-4.18-CL-3832480
LogInit: Engine Version: 4.18.3-3832480+++UE4+Release-4.18
[... 125 lines cut ...]
[2019.01.24-09.25.58:448][ 15]LogMemory: Virtual Memory: 6139.82 MB used,  2919.10 MB free, 134217728.00 MB total
[2019.01.24-09.26.03:491][ 59]LogThreadingWindows: Error: Runnable thread FRecordingThread crashed.
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: === Critical error: ===
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: 
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: Fatal error!
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: 
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: Unhandled Exception: 0xe06d7363
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: 
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: [Callstack] 0x000000008482A388 KERNELBASE.dll!UnknownFunction []
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: [Callstack] 0x00000000462B4722 VCRUNTIME140.dll!UnknownFunction []
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: [Callstack] 0x00000000415BFC62 MSVCP140.dll!UnknownFunction []
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: [Callstack] 0x00000000E510F7B8 AirSimNH.exe!UnrealImageCapture::getSceneCaptureImage() [d:\vso\msresearch\theseus\airsimenvnh\plugins\airsim\source\unrealimagecapture.cpp:40]
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: [Callstack] 0x00000000E510E315 AirSimNH.exe!UnrealImageCapture::getImages() [d:\vso\msresearch\theseus\airsimenvnh\plugins\airsim\source\unrealimagecapture.cpp:30]
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: [Callstack] 0x00000000E51084AF AirSimNH.exe!FRecordingThread::Run() [d:\vso\msresearch\theseus\airsimenvnh\plugins\airsim\source\recording\recordingthread.cpp:91]
[2019.01.24-09.26.03:491][ 59]LogWindows: Error: [Callstack] 0x00000000E26EB2CC AirSimNH.exe!FRunnableThreadWin::Run() []
[2019.01.24-09.26.03:492][ 59]LogWindows: Error: [Callstack] 0x00000000E26DBB13 AirSimNH.exe!FRunnableThreadWin::GuardedRun() []
[... cut ...]
[2019.01.24-09.26.03:492][ 59]LogWindows: Error: [Callstack] 0x0000000087823691 ntdll.dll!UnknownFunction []
[2019.01.24-09.26.03:492][ 59]LogWindows: Error: [Callstack] 0x0000000087823691 ntdll.dll!UnknownFunction []
[2019.01.24-09.26.03:492][ 59]LogWindows: Error: 
[2019.01.24-09.26.03:492][ 59]LogWindows: Error: Crash in runnable thread FRecordingThread
[2019.01.24-09.26.03:513][ 59]LogExit: Executing StaticShutdownAfterError
[2019.01.24-09.26.03:513][ 59]LogWindows: FPlatformMisc::RequestExit(1)
[2019.01.24-09.26.03:513][ 59]Log file closed, 01/24/19 10:26:03
`

madratman, 2019-02-19:
I am closing this as this was a one-off and OP got things working

closed 2019-02-19 by madratman

cacadam89, 2019-04-26:
I am trying something similar here, where I want to set the position of both cameras. In this case I want the ground truth (no noise) and the noisy image from the same location. However, one of the two cameras does not move. 

I attached two screen shots showing all cameras set to the origin (already something is amiss - only one shows the drone's arm which is wrong since both should see the same thing). In the second image I set both cameras to be forward, and here it is even more clear that one of the cameras does not move.

[image]

[image]

Settings with all cameras at origin (i.e. both are set to `"X": 0.00, "Y": 0.00, "Z": 0.00`):
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "scene_gt", "Visible": true},
    {"WindowID": 2, "ImageType": 0, "CameraName": "scene_noisy", "Visible": true}
  ],
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 2,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "scene_gt": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 0.00, "Y": 0.00, "Z": 0.00,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "scene_noisy": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "NoiseSettings": [
            {
              "Enabled": true,
              "ImageType": 0,
              "RandContrib": 0.2,
              "RandSpeed": 100000.0,
              "RandSize": 500.0,
              "RandDensity": 2,
              "HorzWaveContrib":0.03,         
              "HorzWaveStrength": 0.08,
              "HorzWaveVertSize": 1.0,
              "HorzWaveScreenSize": 1.0,
              "HorzNoiseLinesContrib": 1.0,
              "HorzNoiseLinesDensityY": 0.01,
              "HorzNoiseLinesDensityXY": 0.5,
              "HorzDistortionContrib": 1.0,
              "HorzDistortionStrength": 0.002
            }
          ],
          "X": 0.00, "Y": 0.00, "Z": 0.00,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  }
}`

Settings with both cameras shifted forward  (i.e. both are set to `"X": 10.00, "Y": 0.00, "Z": 0.00`):
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "scene_gt", "Visible": true},
    {"WindowID": 2, "ImageType": 0, "CameraName": "scene_noisy", "Visible": true}
  ],
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 2,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "scene_gt": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 10.00, "Y": 0.00, "Z": 0.00,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "scene_noisy": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "NoiseSettings": [
            {
              "Enabled": true,
              "ImageType": 0,
              "RandContrib": 0.2,
              "RandSpeed": 100000.0,
              "RandSize": 500.0,
              "RandDensity": 2,
              "HorzWaveContrib":0.03,         
              "HorzWaveStrength": 0.08,
              "HorzWaveVertSize": 1.0,
              "HorzWaveScreenSize": 1.0,
              "HorzNoiseLinesContrib": 1.0,
              "HorzNoiseLinesDensityY": 0.01,
              "HorzNoiseLinesDensityXY": 0.5,
              "HorzDistortionContrib": 1.0,
              "HorzDistortionStrength": 0.002
            }
          ],
          "X": 10.00, "Y": 0.00, "Z": 0.00,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  }
}`

wzhwzh94, 2020-06-05:
Don't use subWindow 0.
Try:
"SubWindows": [ {"WindowID": 1, "ImageType": 0, "CameraName": "scene_gt", "Visible": true}, {"WindowID": 2, "ImageType": 0, "CameraName": "scene_noisy", "Visible": true} ]